ThingPulse/esp8266-oled-ssd1306 #2: screen misaligned after reflash

tzapu, 2015-12-01:
when flashing the wather code a second time the display is usually misaligned to some random x/y and only recovers after a complete power cycle

i don t know if this could be just my display doing it... got it from banggood

[image]

squix78, 2015-12-05:
I have the same problem with the 4pin I2C version. They are missing a physical #reset line which you would use to start the proper display initialisation. I have been trying to find out if I can do the same with a I2C command init sequence but so far no success...

squix78, 2015-12-05:
I think I fixed it, couldn't reproduce it after the fix. @tzapu can you please check if the issue still occurs? Thx.

tzapu, 2015-12-07:
you are my hero
this has fixed it :+1: 
cheers

closed 2015-12-07 by tzapu

squix78, 2015-12-07:
Cool, thank you for your input and re-testing it!