ThingPulse/esp8266-oled-ssd1306 #3: single frame with no indicators

tzapu, 2015-12-02:
hi,

how would you got about using this great lib for just a single frame without indicators and no switchover animations?

cheers

squix78, 2015-12-02:
Hi

That is actually already implemented. If you don't define the callbacks with setFrameCallbacks(..) there will be no indicators and also no transition. In that case just write your drawing code directly in the loop() method. Dont forget to use display.clear() to empty the display and display.display() to commit the internal buffer to the OLED display memory. Your drawing should be called between the two calls... Can you please let me know if this really worked, then I will close this issue...;-)

tzapu, 2015-12-02:
very nice, just tried it, seems to work fine. maybe add it to the read me?
bangin' job you ve done with this. cheers

closed 2015-12-02 by tzapu

squix78, 2015-12-02:
BTW: added this to the readme:
The Frane Transition functions are a set of functions on top of the basic library. They allow you to easily write frames which will be shifted in regular intervals. The frame animation (including the frame indicators) will only be activated if you define callback functions with setFrameCallacks(..). If no callback methods are defined no indicators will be displayed.

tzapu, 2015-12-02:
cool, that gives everyone a starting point if they don t want the animation (the demo is also animation based)

squix78, 2015-12-02:
good point, maybe I should add more examples;-). Well, sadly its just a hobby, maybe I'll find time...