ThingPulse/esp8266-oled-ssd1306 #4: support for SH1106

tzapu, 2015-12-09:
Hi,
Any chance of supporting the SH1106 as well? 
The only difference seems to be the size of the memory buffer, like described here
https://forum.arduino.cc/index.php?topic=256374.0
>There is only one small difference between SSD1306 and SH1106: The SH1106 controller has an internal RAM of 132x64 pixel. The SSD1306 only has 128x64 pixel.
First problem for standard 128x64 OLEDs with SH1106 controller is: How is the 128x64 window mapped into the 132x64 RAM?
It seems, that the 128x64 OLED is centered in most cases within the 132x64 area, that means pixel (2,0) in ram is pixel (0,0) on the display. 
For u8glib, the update for the SH1106 was very small. I just had to shift the display by 2 pixel. This is more complicated with the Adafruit lib.

Cheers

Zodiac-69, 2016-03-22:
I would also appreciate if you can include SH1106 support

Zodiac-69, 2016-03-23:
I can do testing and debug for you on the ESP8266 - Digidtump Oak -  and the SH1106

SirUli, 2016-05-20:
If you want to change it on your own: http://blog.squix.org/2015/12/esp8266-weather-station-new-version.html#comment-400

FWeinb, 2016-05-29:
I can add support for SH1106 once I get one of those 1.3'' modules. The foundation for this is #41.

hallard, 2016-06-13:
Hi guys,

just to be sure I'm not missing something, SH1106 displays are not yet supported in v3.0.0 ? correct ?
I've got some so if anyone can point me where to do modifications in v3.0.0, I can test and do PR.
I imagine I need to create a new file SH1106Wire.h for example but I think modification are so small that may be some `#ifdef OLED_TYPE_SH1106`in `SSD1306Wire.h` and a new `SH1106.h` lcontaining new define symbol for SH1106 could be easy ?

```arduino
// File SH1106.h
#ifndef SH1106_h
#define SH1106_h
#define OLED_TYPE_SH1106
#include "SSD1306Wire.h"
#endif
```
just let me know

SirUli, 2016-06-13:
Hi @hallard 
currently it is not supported, yes. The changes that i had linked are valid for an older version. Maybe this: https://github.com/rene-mt/esp8266-oled-sh1106 helps to figure out what needs to be changed?

Best Regards, Uli

hallard, 2016-06-13:
Yeah, I remember I already done this for My [ArduiPi OLED](https://github.com/hallard/ArduiPi_OLED/blob/master/ArduiPi_OLED.cpp) for SH1106 support, apparently [one diff](https://github.com/hallard/ArduiPi_OLED/blob/master/ArduiPi_OLED.cpp#l543) on init and other when transferring [data](https://github.com/hallard/ArduiPi_OLED/blob/master/ArduiPi_OLED.cpp#l923) but with this new lib there is some minBoundY and maxBoundY that I'm not sure to understand, and having time to think about this.

squix78, 2016-06-14:
I guess the problem is that neither Fabrice nor me have a display for development. I will order one now but it will take a whlie until it arrives. In the meantime you are more than welcome to create a pull request.

hallard, 2016-06-15:
Should be okay with this [PR](https://github.com/squix78/esp8266-oled-ssd1306/pull/52)

tueddy, 2016-06-26:
I got this one within a few days for 10Euro (shiiping from germany) from Eckstein-Shop:
Waveshare 1.3 inch 128x64 OLED SPI/I2C Interfaces vertical/horizontal pinheader SH1106

So i vote +1
Best regards
Dirk

FWeinb, 2016-06-30:
Released in `3.2.1`

closed 2016-06-30 by FWeinb

tzapu, 2016-09-12:
tested and works great btw