ThingPulse/esp8266-oled-ssd1306 #5: drawString with Umlauts are not working

lighthousebulb, 2015-12-11:
When I use a Umlaut in a display.drawString-Function inside a frameCallback it shows Garbage on the Display.

```
display.drawString(0 + x, 25 + y, "äöüÄÖÜ");
```
gives me something like: 

[image]

Happens with v1.0.1 of the ssd1306 oled library.

squix78, 2015-12-11:
Thank you for reporting. I think I know how to fix the issue. I think characters above ASCII 127 will be encoded as Unicode or other encoding

squix78, 2015-12-12:
Hi @lighthousebulb 
I just added a fix and released a new version of the library. Can you please confirm that the latest release fixes the Umlauts? If yes please close this issue...
Cheers,
Dani

lighthousebulb, 2015-12-12:
Works perfect so far!
Thanks!

closed 2015-12-12 by lighthousebulb