ThingPulse/esp8266-oled-ssd1306 #6: How to add new fonts?

pratikshinde, 2015-12-12:
I have been trying to add new font with The Dot Factory Tool, but rendering is not proper,
How do you add fonts?

squix78, 2015-12-12:
Hi Pratik. I'm currently working on a web app where you can create the fonts. If you cannot wait until it is finished, here is a very hacky version of the future tool in java:
```java
package ch.squix.FontConverter;

import java.awt.Color;
import java.awt.Font;
import java.awt.Graphics;
import java.awt.GraphicsEnvironment;
import java.awt.image.BufferedImage;
import java.io.IOException;
import java.io.PrintStream;
import java.nio.ByteBuffer;
import java.nio.CharBuffer;
import java.nio.charset.Charset;
import java.nio.charset.CharsetDecoder;
import java.nio.charset.CoderResult;

/**
 * Font converter
 * 
 */
public class TtfBitmapFontConverter {
	
	private static final int END_CHAR = 256;
	private static final int START_CHAR = 32;
	private Graphics g;
	private BufferedImage image;

	public TtfBitmapFontConverter(Font font) {
		image = new BufferedImage(100, 100, BufferedImage.TYPE_INT_RGB);
		g = image.getGraphics();
		g.setFont(font);
	}
	
	public void printFontArray(PrintStream writer) {
		String fontName = g.getFont().getFontName() + "_" + getFontStyle() + "_" + g.getFont().getSize();
		writer.println("const char " + fontName + "[] PROGMEM = {");
		writeHexValue(writer, "Width", getMaxCharWidth());
		writeHexValue(writer, "Height", getMaxCharHeight());
		writeHexValue(writer, "First Char", START_CHAR);
		writeHexValue(writer, "Numbers of Chars", END_CHAR-START_CHAR);
		writer.println();
		writer.println("\t// Char Widths:");
		for (int i = START_CHAR; i < END_CHAR; i++) {
			writeHexValue(writer, String.valueOf(i), g.getFontMetrics().charWidth((char)i));			
		}

		
		writer.println();
		writer.println("\t// Font Data:");
		for (int i = START_CHAR; i < END_CHAR; i++) {
			writeCharacterData(writer, (char)i, i == END_CHAR - 1);
		}
		writer.println("};");
	}
	
	private String getFontStyle() {
		Font font = g.getFont();
		if (font.isPlain()) {
			return "Plain";
		} else if (font.isItalic() && font.isBold()) {
			return "ItalicBold";
		} else if (font.isBold()) {
			return "Bold";
		} else if (font.isItalic()) {
			return "Italic";
		}
		return "";
		
	}
[... 118 lines cut ...]
	}
	
	public int getMaxCharHeight() {
		return g.getFontMetrics().getMaxAscent() + g.getFontMetrics().getMaxDescent() + g.getFontMetrics().getLeading();
	}
	
	public Graphics getGraphics() {
		return g;
	}

	public static void main(String[] args) throws InterruptedException, IOException {

		
		TtfBitmapFontConverter app = new TtfBitmapFontConverter(new Font("Arial", Font.PLAIN, 10));
		app.printFontArray(System.out);
		app = new TtfBitmapFontConverter(new Font("Arial", Font.PLAIN, 16));
		app.printFontArray(System.out);
		app = new TtfBitmapFontConverter(new Font("Arial", Font.PLAIN, 24));
		app.printFontArray(System.out);

	}

}

```

juanpintom, 2016-02-03:
Hi! What about convert bitmaps? a Dot Factory profile will be good :P And your webpage too :)

Thanks for the library ;)

somebox, 2016-04-11:
The webpage for the font tool is here: http://oleddisplay.squix.ch
This is noted in the readme, but just in case people end up on this issue via a google search :)

closed 2016-05-29 by FWeinb