ThingPulse/esp8266-oled-ssd1306 #11: Compile Error

carlhako, 2015-12-21:
Hi, Trying to compile the example. Ubuntu 14.04 VM, generic esp-12. 

Arduino: 1.6.6 (Linux), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), ck"

SSD1306Demo:55: error: 'drawFrame1' was not declared in this scope
 void (*frameCallbacks[])(int x, int y) = {drawFrame1, drawFrame2, drawFrame3, drawFrame4};
                                           ^
SSD1306Demo:55: error: 'drawFrame2' was not declared in this scope
 void (*frameCallbacks[])(int x, int y) = {drawFrame1, drawFrame2, drawFrame3, drawFrame4};
                                                       ^
SSD1306Demo:55: error: 'drawFrame3' was not declared in this scope
 void (*frameCallbacks[])(int x, int y) = {drawFrame1, drawFrame2, drawFrame3, drawFrame4};
                                                                   ^
SSD1306Demo:55: error: 'drawFrame4' was not declared in this scope
 void (*frameCallbacks[])(int x, int y) = {drawFrame1, drawFrame2, drawFrame3, drawFrame4};
                                                                               ^
exit status 1
'drawFrame1' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

carlhako, 2015-12-21:
with verbose output

Arduino: 1.6.6 (Linux), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), ck"

/home/esp8266/arduino-1.6.6/arduino-builder -dump-prefs -logger=machine -hardware "/home/esp8266/arduino-1.6.6/hardware" -hardware "/home/esp8266/.arduino15/packages" -tools "/home/esp8266/arduino-1.6.6/tools-builder" -tools "/home/esp8266/arduino-1.6.6/hardware/tools/avr" -tools "/home/esp8266/.arduino15/packages" -built-in-libraries "/home/esp8266/arduino-1.6.6/libraries" -libraries "/home/esp8266/Arduino/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=ck -ide-version=10606 -build-path "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display/examples/SSD1306Demo/SSD1306Demo.ino"
/home/esp8266/arduino-1.6.6/arduino-builder -compile -logger=machine -hardware "/home/esp8266/arduino-1.6.6/hardware" -hardware "/home/esp8266/.arduino15/packages" -tools "/home/esp8266/arduino-1.6.6/tools-builder" -tools "/home/esp8266/arduino-1.6.6/hardware/tools/avr" -tools "/home/esp8266/.arduino15/packages" -built-in-libraries "/home/esp8266/arduino-1.6.6/libraries" -libraries "/home/esp8266/Arduino/libraries" -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=ck -ide-version=10606 -build-path "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display/examples/SSD1306Demo/SSD1306Demo.ino"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp/sketch/SSD1306Demo.ino.cpp"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "-I/home/esp8266/Arduino/libraries/ArduinoJson/src" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display" "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp/sketch/SSD1306Demo.ino.cpp"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "-I/home/esp8266/Arduino/libraries/ArduinoJson/src" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display" "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp/sketch/SSD1306Demo.ino.cpp"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "-I/home/esp8266/Arduino/libraries/ArduinoJson/src" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display" "/tmp/buildec367a1ae5689a511ad63e8f23e47d3f.tmp/sketch/SSD1306Demo.ino.cpp"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display" "-I/home/esp8266/Arduino/libraries/ArduinoJson/src" "/home/esp8266/Arduino/libraries/ArduinoJson/src/Arduino/Print.cpp"
"/home/esp8266/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -w -x c++ -M -MG -MP -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266  -DESP8266 "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/cores/esp8266" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/variants/generic" "-I/home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire" "-I/home/esp8266/Arduino/libraries/ESP826 [... cut ...]
                                                                               ^
Using library ArduinoJson at version 5.0.7 in folder: /home/esp8266/Arduino/libraries/ArduinoJson 
Using library Wire at version 1.0 in folder: /home/esp8266/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/Wire 
Using library ESP8266_Oled_Driver_for_SSD1306_display at version 1.0.2 in folder: /home/esp8266/Arduino/libraries/ESP8266_Oled_Driver_for_SSD1306_display 
exit status 1
'drawFrame1' was not declared in this scope

carlhako, 2015-12-21:
I fixed this by moving like 55 to 69 just before   display.setFrameCallbacks(frameCount, frameCallbacks);

carlhako, 2015-12-27:
I finally got to the bottom of the problem. Most ESP8266 sketches fail to compile and run properly using arduino ide 1.6.7 I reverted back to 1.6.5 now everything works perfectly including the demo.

This issue can now be closed.

FWeinb, 2016-05-29:
This is a Bug in the Arduino IDE not generating the needed prototypes automatically. The examples will be update in 3.0.0.

closed 2016-06-03 by FWeinb