ThingPulse/esp8266-oled-ssd1306 #21: feature request, turn off indicator

matricies, 2016-01-21:
A useful addition would be allowing 'NONE' to be included in the following setting which would prevent display of the any indicators. This would leave more space for other screen content in frames, if needed.

    // TOP, LEFT, BOTTOM, RIGHT, NONE
    ui.setIndicatorPosition(NONE);

As an extension of the above, allow a list of frame numbers which should include indicators, as in: 

    // TOP, LEFT, BOTTOM, RIGHT, NONE. [int frame1, int frame2, ... int framen]
    ui.setIndicatorPosition(BOTTOM, 1, 3, 4);

The above example would show the frame indicators at the bottom for frames 1, 3, and 4 only.
if only TOP, LEFT, BOTTOM or RIGHT were included, the frame indicators would be shown on all frames. if NONE were included, there would be no frame indicators.

FWeinb, 2016-01-22:
This is handelt in #16 via [these function](https://github.com/FWeinb/esp8266-oled-ssd1306/blob/improve-performance-3.0.0/SSD1306Ui.h#L199-L212) you can basically call: 
`ui->disableIndicator()` in a FrameCallback. This will even animate the indicator out, and back in. If nothing is called the indicator is enabled by default.

```c++
void drawFrame(SSD1306 *display, SSD1306UiState* state, int16_t x, int16_t y){
  ui.disableIndicator();
  // draw on the full display for this frame
}
```

matricies, 2016-01-26:
Thanks, I got the newer version and it works perfectly.

This is a very nice library and gets amazing results from such a modest OLED display.

closed 2016-01-26 by matricies