ThingPulse/esp8266-oled-ssd1306 #23: Problems with ESP8266 201

stex2005, 2016-02-23:
I'm trying to configure a ESP 201 as a DHT sensor and I want it to show values on an OLED.
I'm having troubles with it: I can't make the ssd1306 work with the simplest example code provided with the library.

First, is this library compatible with the 201 version?

What are the most suitable pins for I2C interface?

Thank you

dreamer2, 2016-03-12:
it works for me
201 and ssd1306 128x64, 4,5 pins
i have another problem - display initializes only on esp reset or after upload
cold poweron dont show anything

petamoravec, 2016-03-18:
Could you, guys, please post your working code for my inspiration? Also I'm interested in your HW connection ESP201 with DHT and ssd1306 display...
Thanks, P.

FWeinb, 2016-06-03:
When you still have problem feel free to reopen this issue.

closed 2016-06-03 by FWeinb