ThingPulse/esp8266-oled-ssd1306 #28: contributing & naming conventions

hixfield, 2016-04-09:
I am an experienced C++ programmer (embedded). I love the work you have done and would like to participate. However, I noticed that the naming conventions are not ideal. Would you allow me to update the naming of the variables? I also have a few other updates in mind after this.

squix78, 2016-04-09:
Hi Hixfield. You are more than welcome to help. My problem is that I currently have other priorities in my spare time projects and don't have much time to maintain this project. @FWeinb already started a V3.0.0 branch and it probably makes a lot of sense to continue work on that branch, also to avoid breaking downstream dependencies... What do you think?

hixfield, 2016-04-09:
That sounds good to me. @FWeinb are you open for collaboration?

FWeinb, 2016-04-10:
Sure. I made an PR #16 Would really like to clean that up. The PR is against the dev branch for 3.0. Would be good to merge that one first and you can than pick it up and modify as you like.

hixfield, 2016-04-10:
@squix78 would you agree? How do you see the rights issue? Should we get the required rights from you?

squix78, 2016-04-10:
Done. Please check if you have now commit permissions.

FWeinb, 2016-04-10:
Thanks. I can commit. I would now merge the Outstanding PR and @hixfield could start refactoring from there.

hixfield, 2016-04-10:
Sounds like a plan! Let me know when I can pull it and from where :)

squix78, 2016-04-10:
Great. Thank you guys in advance!

FWeinb, 2016-04-10:
Just merged it to the `dev-branch-3.0.0` branch. You can now checkout that branch and work on the code.

FWeinb, 2016-04-10:
I would say you crate a new branch of of the `dev-branch-3.0.0` and than do a PR against the dev-branch when you are done. Me and @squix78 can than review your changes and when everything looks good we merge it.

hixfield, 2016-04-10:
ok that sound good

hixfield, 2016-04-10:
By the way this is the naming convention I have been using for many years, everybody ok with this?
Variable names context prefix:
m_ : member variable
g_ : global variable

Variable names type prefix:
sz : zero terminated character array string
p : pointer
n : signed number
u : unsinged number
c : class

camelCase for variables, functions
PascalCase for objects, types
constants or defines in UPPERCASE

Example:

const in MAXSIZE = 255
#define MAXSIZESTRING 255

class CMyClass : CSomeRootClass {
private :
char m_szSearch[MAXSIZESTRING];
int m_nCharIndex;
char * m_pIndexPointer;

public:

int add(int nNumber1, int nNumber2);
}

FWeinb, 2016-04-10:
As long as we maintain the public api where possible I am fine with that. I would not rename the SSD1306 class just to conform with this naming schema.

closed 2016-06-03 by FWeinb