ThingPulse/esp8266-oled-ssd1306 #29: 128x32 not working

RetepV, 2016-04-12:
Hi, I have a 128x32 display, got it to work, but noticed that there were issues. I fixed them (on the main branch code), and was wondering if you would like me to commit those changes to the main branch.

It's a quick fix, not a cleanup like the 3.0 branch. So I'd recommend to put these quick fixes in the main branch. And then I could test and (if necessary) fix the 3.0 branch as well.

What do you think?

RetepV, 2016-04-15:
I have a few local branches with my fixes, but I'm not authorized to create remote branches in your space. So I can't create pull requests. What do you think, can I contribute? ;)

phil-schneider, 2016-04-29:
Fork this repo to your "own" repository.
There you can do whatever you want.
You should create a bugfix/29 Branch and commit your changes.
Then create a pullrequest of your bugfix branch and the destination should be this original repo.
Thats how it normally works.

closed 2016-06-03 by FWeinb