ThingPulse/esp8266-oled-ssd1306 #33: Simple example...

kmitchel, 2016-04-25:
Please provide a simple/minimalist example.

squix78, 2016-05-19:
I understand your request for a simple example. I guess you are mostly interested in an example without the frames? But I can't follow on the keywords. Do you mind to explain? Do you mean the keywords in the library.json? From which platform did you try to use the library? Arduino IDE, Platformio?

kmitchel, 2016-05-19:
You should make this library easier to find--some how. I messed around with a lot of stuff that didn't work, until I came across your stuff by accident. 

So some search engine optimization stuff...anyone who searches for esp8266 and oled should come here first.

Thank you for the code.

closed 2016-05-19 by kmitchel