ThingPulse/esp8266-oled-ssd1306 #36: Add progress bar function to UI class

squix78, 2016-05-20:
Add a function that simply lets the user draw a progress bar.

FWeinb, 2016-05-20:
I have a rough function for this:

```       
void drawProgressBar(uin8t_t x, uint8_t y, uint8_t height, uint_t width, uint8_t progress) { 
          display->drawRect(x, y, width, height);
          display->fillRect(x + 2, y + 2, ((width - 3) * ((float)progress / 100)), height - 3);
}
```

squix78, 2016-05-20:
I have seen that, but I now have implemented it and (I think) it looks prettier than just the squared progress bar, more "IOS like". I first had it in the UI class, but then moved it to the core class, it felt better there. What do you think?

FWeinb, 2016-05-21:
I think It should ne on the UI class.

FWeinb, 2016-05-29:
This was done in 57e2ff322a306467a2bbed2a395edba9620da60f

closed 2016-05-29 by FWeinb