ThingPulse/esp8266-oled-ssd1306 #46: brzo_i2c not included

HanVertegaal, 2016-06-04:
Installed in PlatformIO and trying to use SSD1306Brzo.
I get an error message indicating that <brzo_i2c.h> cannot be included (from SSD1306Brzo.h).
Brzo i2c is not a separate library for PlatformIO. How do I best install that? Or should it be included in the ESP8266_SSD1306 library?

FWeinb, 2016-06-04:
No brzo_i2c will not be included here. You have to download the code separately from [this](https://github.com/pasko-zh/brzo_i2c) repository.  

I hope that brzo_i2c will be available via platformio in the future.

closed 2016-06-04 by HanVertegaal

squix78, 2016-06-04:
FYI: I added a pull request in brzo repository for including the library files. If Pasko accepts the library will soon be available in the platformio and arduino ide library manager:
https://github.com/pasko-zh/brzo_i2c/pull/3