ThingPulse/esp8266-oled-ssd1306 #47: Font Converter with Dialog font - 2 issues

bHogan2, 2016-06-04:
I made the following font - Dialog, Bold, 10pts, >=3.0

The first problem is that the array name comes out as:
const char Dialog.bold_Bold_10[] PROGMEM = {
(The dot after "Dialog" causes a compile error.)

Changing the line to "Dialog_Bold_10" allows it to compile.

Second problem:
This font worked fine when I made it for v2.0 of the SSD1306 lib.
However when made for the 3.0 lib, it fails to display an uppercase 'M'.
(Widest letter?)

Thanks for a great lib!

FWIW the lib seems to work with ATmega chips if you make the following change in the init()
#ifdef ESP8266 // wire args only if ESP
  Wire.begin(mySda, mySdc);
#else
Wire.begin();
#endif

FWeinb, 2016-06-05:
I just fixed the rendering issue of the uppercase 'M'. We will release this in `3.0.1`. 

@squix78 could you have a look at the font naming issue in the web app?

squix78, 2016-06-20:
Just fixed the issue with the dot and also removed the double placement of the font style in the variable name. @bHogan2 can you please verify that the font converter works now as expected?

bHogan2, 2016-06-20:
Thanks. I can confirm that the problem is fixed with **Dialog, bold, 10pt, >=3.0 font**. I may have seen it  with other fonts, but I don't recall which. The uppercase "M" still doesn't display but my understanding is that it will be fixed in the next release of the lib.
Thanks again for the best ESP/OLED lib, and all the best.
John

closed 2016-06-20 by bHogan2