ThingPulse/esp8266-oled-ssd1306 #49: This new version will cause your sketches to crash

RogerSchaefer, 2016-06-05:
The new naming conventions in Ver 3.0 in combination with the Arduino automatic Library Manager will cause all sketches using esp8266-oled-ssd1306 Ver 2.0 to crash.

You should rename your library, call the new one Ver 1.0 and restore esp8266-oled-ssd1306 to Ver 2.0.2 

Don't close this issue without doing something to warn users of this problem.

Roger

FWeinb, 2016-06-05:
The Arduino Library Manager will not update any library without user interaction. I added a note about the 3.0.0 update to the [README.md](https://github.com/squix78/esp8266-oled-ssd1306/blob/master/README.md). 

Thanks for reporting.

closed 2016-06-05 by FWeinb

squix78, 2016-06-05:
Hi Roger. I just added a warning to the Weather Station code together with a link to the oled library upgrade instructions. When we started the development on the 3.0.0 version I was aware that this would cause grief with the users of the library. But if the Arduino IDE had a halfway decent dependency management system you would be able to pin the 2.x.x version to your sketch. Your suggestion to rename the library for the new version is understandable but in my opinion not the right way to go. It would cause even more chaos than a version update.
When you say "causes sketches to crash" do you mean they do no longer compile or that they compile but crash on the device? I would expect the first, the second would very much surprise me...

RogerSchaefer, 2016-06-05:
Yes it means it will no longer compile.

If folks added new libraries manually as in the past your warnings would be good enough but when Arduino Library Manager says there is a new version most will just click on install.

Maybe they won’t even try to load a sketch that uses “esp8266-oled-ssd1306” for weeks or months later.  A sketch that worked perfectly in the past will now no longer compile and the user will have no idea why.

Roger

From: Daniel Eichhorn 
Sent: Sunday, June 05, 2016 11:29 AM
To: squix78/esp8266-oled-ssd1306  
Cc: RogerSchaefer ; Author 
Subject: Re: [squix78/esp8266-oled-ssd1306] This new version will cause your sketches to crash (#49)

Hi Roger. I just added a warning to the Weather Station code together with a link to the oled library upgrade instructions. When we started the development on the 3.0.0 version I was aware that this would cause grief with the users of the library. But if the Arduino IDE had a halfway decent dependency management system you would be able to pin the 2.x.x version to your sketch. Your suggestion to rename the library for the new version is understandable but in my opinion not the right way to go. It would cause even more chaos than a version update.
When you say "causes sketches to crash" do you mean they do no longer compile or that they compile but crash on the device? I would expect the first, the second would very much surprise me...