ThingPulse/esp8266-oled-ssd1306 #51: disableIndicator() doesn't disable indicators

squix78, 2016-06-12:
Following sequence the indicators are still displayed:

  ui.setTargetFPS(30);
  ui.disableIndicator(); <- no effect
  ui.setFrameAnimation(SLIDE_LEFT);
  ui.setFrames(frames, numberOfFrames);
  ui.setOverlays(overlays, numberOfOverlays);
  ui.init();

FWeinb, 2016-06-12:
You have to call this function inside your frame drawing function for which you want to disable the Indicator. This was added to allow some kind of full screen frames.

squix78, 2016-06-12:
I see. My use case was to completely disable it and display 1/4, 2/4 etc. instead. My workaround was to set an empty icon to active and inactive. Maybe we should add a comment explaining the current behaviour?

hallard, 2016-06-14:
Correct I spent some time to understand this point, I must use in frame drawing callbacks ;-)
In the meantime, same for to `flipScreenVertically()` if done after `display.init()` need to be called again after `ui.init()`

FWeinb, 2016-07-09:
@hallard 
`ui.init()` calles `display.init()` so you don't have to call them both. The initialisation will send the default screen orientation so the is working as intended. We should probably document this better. 

Back to the topic, I would suggest adding  `ui.disableAllIndicators();` for disabling indicators at the UI level and not the frame level.

hallard, 2016-07-09:
thanks for the explanation @FWeinb 
The deal is I'm not using frame during setup (but still use OLED to display information) that why UI is called only at setup end  and begin() at the beginning of setup ;-)

I'll try the global `ui.disableAllIndicators();` thanks

By the way, nothing with this subject but more with documentation, when using `ui.setTimePerFrame(time)`  and `ui.setTimePerTransition(time)` what's the unit, not sure it's ms but kind of tick not ?

If I want each frame duration to 1 second do I need to call `ui.setTimePerFrame(1000)`?

thanks

FWeinb, 2016-07-09:
The function `disableAllIndicators()` dons't exist yet. We need to add it in a next version. 
Time based function (`ui.setTimePerFrame(time)`and `ui.setTimePerTransition`) are in seconds.

FWeinb, 2016-07-24:
Fixed in #71

PR #71 (2016-07-24): Make it possible to enable/disable all indicators
This fixes #51 by adding `disableAllIndicators()` and the corresponding `enableAllIndicators()`.
closed 2016-07-24 by FWeinb