ThingPulse/esp8266-oled-ssd1306 #53: Internal image format

kaoz3000, 2016-06-18:
Hi,
what a great, fast Library! I Just have fun with some random graphics.
I just wonder what the drawFastImage() is about.
Where can I find a converter for this datatype:

> Draw a bitmap in the internal image format
> `void drawFastImage(int16_t x, int16_t y, int16_t width, int16_t height, const char *image);`

Thank you!
Felix

squix78, 2016-06-20:
Hi Felix

I guess we'll have to ask @FWeinb for a definitve answer but I think it is a straight forward bitmapping, and the method is used by the string drawing methods as well to draw the inactive and active symbols for the carousel. Looking at the definition of the active and inactive symbols here: https://github.com/squix78/esp8266-oled-ssd1306/blob/0b88d0cff9fbc4dba7dfa4f8521520c0445b9555/examples/SSD1306UiDemo/images.h
tells me that it should be fairly easy to draw something just by setting bits in a byte. I'm not sure what happens when the image width is not a multiple of eight, if the extra bits are ignored or continued on the next line or if this even isn't allowed. I'm sure Fabrice doesn't mind to enlighten us but he is currently busy with exams... So please stay tuned!

FWeinb, 2016-06-20:
Just had a minute here, the `drawFastImage` is used to speed up the rendering as it allows us to pretty much copy the image onto the display. You can thing of the display a a big memory blob somewhere in RAM. Each char in this buffer corresponds to 8 pixels (imagine a char represented in base 2. e.g. `00000001`) vertically.  So for each 128 chars we can draw 8 pixels vertically and 128 horizontally. The only thing you have to keep in mind when [defining ](https://github.com/squix78/esp8266-oled-ssd1306/blob/0b88d0cff9fbc4dba7dfa4f8521520c0445b9555/examples/SSD1306UiDemo/images.h#L30-L39) this is that the actual image will be rotated 90 deg clockwise. 
 
You can draw any number of pixels, the draw function will handle it.

closed 2016-06-25 by FWeinb

doronby, 2017-12-18:
Did you ever get to document better how to make from BMP the array to use the DrawFastImage function?

Chi5487, 2017-12-27:
Long story short:
_At the moment I'm sick... so brain is clouded, sorry for that... and English is not my  mother tongue as well. I'm just a coder, I have nothing to do with this... Just trying to help you..._

Because of the weird 90 degrees rotation, you are limited to the max height of the image as width (so, for a 128x64 screen, your image will be 64x64pix MAX!) 

I did the test (used it on NodeMCU v3) and the following/procedure setup will be working:
- First, create your image in photoshop or whatever... (black pixels will be lit-up on the OLED)
- Crop to 64x64  (you can combine multiple croped images in the end...)
- I think that the image size must be a multiple from 8, I'm not sure...
- Rotate it CCW 90 degrees, and then.. ooo boy … 
- mirror it (flip vertical, I think, you will figure it out)… 
- Save as whatever pref BMP, TIFF (=lossless)
- Upload to https://convertio.co/image-converter/ and convert to image/XBM
- So the clue is XBM !!!
- Download the result, past the array into your code … (you can or ignore PROGMEM)

And then do something like this: 
` display.drawFastImage( 0, 0, 64, 64, Image1);`
    
If you want to add an extra image (second part of the 128*64) to the right of it then do it as follow:
`display.drawFastImage(64, 0, 64, 64, image2); `
    
The top-left corner is 0,0 if you use display.flipScreenVertically();
I did this, so that I have a normal landscape 128x64 screen... nothing more....

Don’t shoot me, I just figured it out that this was just working… just a really bad bad bad bad function! 

Note: forget LCD assistant, it’s not compatible: for each row, the data is totally reversed. And it will rotate 90 degrees as well… just ignore LCD assistant, lost a lot of time to figure that one out...

I'll try to follow this up the next hours/days...
Kind regards,
Chi

Joennuh, 2018-10-24:
@Chi5487 Thank you very much for your post! I was struggling with LCD assistant but only got garbage images. With your instructions I managed it to get images on my display! :-)
The only thing I had to change was
'const char theImage[] PROGMEM = {'
into:
'const uint8_t theImage[] PROGMEM = {'
because of this error:
'invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]'