ThingPulse/esp8266-oled-ssd1306 #54: Best Practice to go to a transition on event ?

hallard, 2016-06-24:
Hi there, 
Sorry, it's more a question then an issue. If there is other channel for this type of request, just let me know.

I'm using UI with automatic transition (4 callbacks) with great success, but in some case (external event) I need to go immediately to a specific transition whatever where I am (for example I'm in # 1 and I want to go immediately to # 3)
Then let the others continue as usual -> # 4, # 1, # 2, # 3 ...
I do not need to go back the one "interrupted" just continue. 
What's the best practice to achieve this ?
Thanks for your help

FWeinb, 2016-06-24:
There is currently no API in place to allow to transition to an arbitrary frame other then the `next` or `previous` on using `nextFrame()` and `previousFrame()`. We should think of adding API to:

* immediately switch the frame using e.g. `switchToFrame(int frame)`.
* transition to a arbitrary frame using e.g. `transitionToFrame(int frame)`.

FWeinb, 2016-06-24:
I implemented the described function in #55. When @squix78 has some time to review the PR we can release it in `3.1.x`.

closed 2016-06-24 by FWeinb