ThingPulse/esp8266-oled-ssd1306 #56: Non Latin Characters

scropion86, 2016-06-25:
Hi,

hi i tried to send some Non Latin Characters (Arabic letters) and nothing was displayed on the screen.
is this a limitation of Added font , or a library limitation.

thanks

FWeinb, 2016-06-26:
A font definition can maximally contain 255 chars. So we can display the extended ascii set of characters. Arabic letters are not in this set so we cant display them. 

One workaround for this is to generate a font definition that consists only of arabic letters and use a little helper function to shift the arabic letters into chars in the 0.254 range. The only caveat would be that you could not mix Arabic and non Arabic letters in the same string.

scropion86, 2016-06-27:
thanks i will test it and update you , and you are correct Arabic is RTL language.
you also can test it with this Arabic `Hello World` is `أهلا بالعالم`
and it should be appearing the same in that comment.

FWeinb, 2016-06-27:
I updated the font to be 16px big and added a reverse function to correctly print the string. I updated the demo [here](https://gist.github.com/FWeinb/0a81620fdf7762d046edb98bf8d5fd2d)

scropion86, 2016-06-27:
thanks for the Demo , i test the last Demo you provided but as i told you the letters comes aparts and not continuous in words.
Arabic is little bit complicated .

[image]

FWeinb, 2016-06-27:
I noticed that but I can't to much about it because I don't understand the rules underpinning the arabic language. I have no idea if there is a simple rule set when to change UTF8 chars und what combination etc.

I am afraid you might be on your own here. If you have any technical questions, feel free to ask. I will try to help as good as I can.

scropion86, 2016-06-27:
thanks a lot for your time and effort,i will keep searching if i found a solution i will let you know.

closed 2016-06-27 by scropion86

raphikWasHere, 2024-05-28:
Sorry to reopen such an old post but... by chance I found a sketch for Arduino to write Persian on an SSD1306 OLED display. As far as I know Persian uses the arabic alphabet so probably it is also useful for writting Arabic. Besides I have a clue, I don't really know who generated the code, but it certainly wasn't me. The original code didn't work properly but I liked it so much that I took the challenge of making it work. Therefore, my only contribution has been debugging it a little wee. Four cases are considered to decide how a letter is written, depending on the position of the letter: isolated, at the beginning, in the middle or tailing the word. I found it a magnificent piece of software. Anyone can try both codes at:
Original code: https://wokwi.com/projects/372442890656329729
Debugged (and working) code: https://wokwi.com/projects/399060267330575361

Best regards.

raphikWasHere, 2024-05-28:
Clues led me to https://github.com/idreamsi/arduino-persian-reshaper as the maker of the sketch. Or am I wrong?