ThingPulse/esp8266-oled-ssd1306 #58: How to use ui.switchToFrame();

hallard, 2016-07-05:
Guys,

I spent some time to test my request to be able to switch to a specific frame (for for coding it by the way). 
I'm trying to use switchToFrame without success but may by I'm doing wrong, here below the code (only part of)

```arduino
void loop() {

  // bla bla check RF packet received

  if (packet_received) {
    // display immediately data received
    ui.switchToFrame(RF_FRAME_INDEX);
    ui.update();

    // Here I'm doing approx 2 second long stuff blocking
    Post_Data();
  }

  ui.update();

  // Handle Web server
  web_server.handleClient();

  // Handle OTA
  ArduinoOTA.handle();

}
```

Frame are scrolling OK but it never switch to the correct one (it does nothing) when I call `switchToFrame()`
If there any example on how do use it ?

Thanks

FWeinb, 2016-07-05:
I have no idea how this could happen but it looks like [there](https://github.com/squix78/esp8266-oled-ssd1306/blob/master/OLEDDisplayUi.cpp#L154) is an error in this condition. It should be: 

```c
if (frame >= this->frameCount || frame == this->state.currentFrame) return;
```

Could you test these changes in your local copy and confirm my guess. I have really no idea why the code was working for me.

hallard, 2016-07-05:
@FWeinb 
You got it it was a bug works with your suggestion. Note that same suggestion need to be done for `transitionToFrame()` also (same bug).

BTW, if we're on the current (and good) frame display, nothing is done (this is expected) but worth resting tick to restart the frame timer. Just because if we're at the end of frame timer we could ask to go to this frame but as we're on it, it does nothing but we could go to the next frame very quickly.

FWeinb, 2016-07-06:
Thanks for confirming, I will create a PR to fix it.

commit f373f110 (2016-07-09): Fixes #58
closed 2016-07-09 by squix78