ThingPulse/esp8266-oled-ssd1306 #59: OverLays are written on top of each other

nicof38, 2016-07-05:
While using the SSD1306UiDemo, I've modified this one in order to have:

`void msOverlay1(OLEDDisplay *display, OLEDDisplayUiState* state) {
  display->setTextAlignment(TEXT_ALIGN_RIGHT);
  display->setFont(ArialMT_Plain_10);
  display->drawString(128, 0, String(millis()));
}

void msOverlay2(OLEDDisplay *display, OLEDDisplayUiState* state) {
  display->setTextAlignment(TEXT_ALIGN_RIGHT);
  display->setFont(ArialMT_Plain_10);
  display->drawString(128, 0, String("TESTTESTTEST"));
}`

Then:

`// Overlays are statically drawn on top of a frame eg. a clock
OverlayCallback overlays[] = { msOverlay1,msOverlay2 };
int overlaysCount = 2;`

Running the example with these modifications, and you can see that "TESTTESTTEST" is written on top of the current milliseconds.

FWeinb, 2016-07-05:
That is the expected behaviour. What is your expectations with doing this? A overlay is a way to define frame independent data on the display (like a clock). If you draw to the same pixels they sure overwrite what is currently there.

nicof38, 2016-07-05:
So in this case what is the aim of having multiple overlay defined?
It does not work the same way as frames? Frames does not overwrite on each
others so I was expecting to have the same behavior for multiple overlays.

One overlay should only overlay frames but I would not expect each overlay
to overlay any others.

Anyway if it is the way it is designed, how should we use the multiple
overlays callback in order to have something readable?

Thanks

FWeinb, 2016-07-05:
In an overlay you can draw anywhere on the display. You can do something like this:
```
void msOverlay2(OLEDDisplay display, OLEDDisplayUiState state) {
display->setTextAlignment(TEXT_ALIGN_LEFT);
display->setFont(ArialMT_Plain_10);
display->drawString(0, 0, String("TESTTESTTEST"));
}
```

That would draw the string "TESTTESTTEST" in the upper left corner

nicof38, 2016-07-05:
Yes, this is what I've done next. Anyway still surprised that 2 overlays
callback can overwrite on each others whereas frames clear everything
between.

What should be the best option to update information on an overlay area
(with different information type, not a timestamp for example)? Should I
add a third overlay in order to clear the selected area ? Should i use an
empty graphic to do so ?

FWeinb, 2016-07-06:
Sorry but I don't understand what you would like to archive. Maybe @squix78 can help? 
Overlays are information that are rendered always (even in a frame transition) so they can be used for status indicators like wifi signals or time, we choose to implement the ability to add more overlays so one can separate different aspects of the ui (not everyone want's to e.g. display an wifi signal indicator).

nicof38, 2016-07-09:
OK, so based on your latest explanation, I think I get it.

Adding more overlays does not behave like adding multiple frames. Every defined overlays are rendered all the time, and that is why they can interfere on each others. 

Maybe you should add the way is has to be used with multiple overlays in the UI demo code ?

FWeinb, 2016-07-15:
Yeah, we could really improve on example code for this library.

closed 2016-07-15 by FWeinb

hallard, 2016-07-16:
If you need some example, I wrote this one to test my boards, it use OLED (of course), the UI to display informations such as Wifi Networks, IC2 devices scanned, drive a WS2812 LED and manage progress bar and WS128 LED with OTA.

Full source code on my (github)[https://github.com/hallard/Scan-I2C-WiFi] and a small video to see in action