ThingPulse/esp8266-oled-ssd1306 #66: drawRect and fillRect produce rects of different size

igrr, 2016-07-10:
Currently the code in OLEDDisplay::drawRect is as follows:
```c++
  drawHorizontalLine(x, y, width);
  drawVerticalLine(x, y, height);
  drawVerticalLine(x + width, y, height);
  drawHorizontalLine(x, y + height, width);
```
Compared to `fillRect` function, this draws a rectangle which is 1 pixel wider and 1 pixel higher. Also it lacks a pixel at the lower-right corner. 

[image]

If I change that to `x + width - 1` and `y + height - 1`, i get same size as for `fillRect`.
I'm not sure if such a "fix" qualifies as a change in API though :)

And by the way, why is `fillRect` using `drawHorizontalLine`? Isn't `drawVerticalLine` faster because it can set up to 8 pixels at a time?

FWeinb, 2016-07-10:
Thanks for reporting, this won't qualify as a change in API because this is clearly a bug in the code. Good idea to improve `fillRect` by using `drawVerticalLine`!

FWeinb, 2016-07-24:
Thanks for reporting. We decided to release this in `4.0.0` with some other fixes that could possible break existing code.

closed 2016-07-24 by FWeinb