ThingPulse/esp8266-oled-ssd1306 #68: Arduino Uno - Build Error

ReVl27, 2016-07-16:
Hi, i'm using esp8266-oled-ssd1306 library for may project.
But while loading the sketch to the Arduino, I got the following error?

error: '_min' was not declared in this scope
 minBoundY = _min(minBoundY, y);

The same happens to _max?

Could you explain to me how to get around this problem, please?
Thank you!

FWeinb, 2016-07-16:
What version of esp8266 arduino are you using? These macros are defined [here](https://github.com/esp8266/Arduino/blob/a14ac2cbdd981956ee933f101592af493fee1a07/cores/esp8266/Arduino.h#L256). We are using _min/_max because there are some issues with the normal min/max macros.

ReVl27, 2016-07-17:
Hi FWeinb,
I'm using version 3.2.3, which I download from GitHub.
Well, it looks like i'm missing the file "arduino.h".
The fact is that I couldn't find such file in the extracted folder either.

FWeinb, 2016-07-17:
This is a library that should be used together with [ESP8266 Arduino](https://github.com/esp8266/Arduino) and the Arduino IDE or [platformio](platformio.org) you should first install the ESP8266 Arduino platform and after that use the library install in the IDE to install this library.

squix78, 2016-07-17:
Just to be clear: are you using the library with an ESP8266 or with an Arduino Uno as indicated in the title of this issue? This library is for ESP8266 target platform only. If you are using standard Arduino (ATMEGA) device then there are other libraries available

ReVl27, 2016-07-17:
@FWeinb & Squix78.
Yes, I do use a standard Arduino (Atmega) at the time. Hence cause this problem.
Thank you guys for making it clear!

closed 2016-07-17 by ReVl27