ThingPulse/esp8266-oled-ssd1306 #73: Support for 64x48 OLED ?

erkobg, 2016-07-25:
Hello,
It will be very helpful if you add support for 64x48 OLED like the official display for Wemos :+1: http://www.wemos.cc/Products/oled_shield.html

Here they did almost the same : 
https://github.com/mcauser/Adafruit_SSD1306/tree/esp8266-64x48

Regards!

Testato, 2016-08-01:
+1

64x48 pixel is a very common display, have you plan for support it ?

FWeinb, 2016-08-02:
I would like to support it but I don't have one yet so I can't start working on that.

hallard, 2016-08-05:
I ordered some last week, as soon as I got them I may take a look and do a PR.
@FWeinb is there any tip should I know for being able to change resolution on library, such like current 128x64 resolution hard coded somewhere? may I just change then and OLED init code ?

FWeinb, 2016-08-11:
To get this working you need to change the defines [here](https://github.com/squix78/esp8266-oled-ssd1306/blob/master/OLEDDisplay.h#L47-L48) and introduce a custom constructor/subclass of each driver to allow for selection of the width/height. Than you need to introduce local member variables to contain the width/height in the OLEDDisplay class and change every access to the defines to use these new local variables.

LimaLima, 2016-09-23:
Being a lazy bastard, I took this lib and instead of adding support for different display, I just made it work with 128x32.
Then to be a complete *hole, I took the latest version, dropped changed files over it instead of merging.
But it might be still a useful starting point for your own changes - feel free to see [diff](https://github.com/LimaLima/esp8266-oled-ssd1306/commit/67cd2bb37e6e60359dc52100afd9ef661fbdf6ae) and find what's needed to make other geometries work.

trepidacious, 2016-12-21:
I've submitted a pull request https://github.com/squix78/esp8266-oled-ssd1306/pull/97 for changes to allow use of either 128x64 or 128x32 displays based on LimaLima's changes (thanks for working out the register values, I don't know what they do, but they work for me!)

artkeller, 2017-01-07:
@Fabrice: to understand the tiny 64x48 display a bit better, refer to the following stuff

// cf. [https://www.wemos.cc/product/oled-shield.html](https://www.wemos.cc/product/oled-shield.html)
// cf. [https://macsbug.files.wordpress.com/2016/04/zahyou_s.png?w=700](https://macsbug.files.wordpress.com/2016/04/zahyou_s.png?w=700)
// cf. [https://macsbug.wordpress.com/2016/04/01/using-the-wemos-i2c-oled-64x48/](https://macsbug.wordpress.com/2016/04/01/using-the-wemos-i2c-oled-64x48/)

Sorry - the last link goes to a japanese site, but it can be understood pretty good due to the well documented code examples in english, good links and nice pictures... (I like this stuff)

BTW: Thanks for the New Year stuff :-)

Thomas

tofrnr, 2018-02-13:
I fixed thet issue for a OLED 64x48:
```
/*
 * for OLED 64x48:
 * x-offset=32
 * y-offset=16
 * 
 * exchange  in OLEDDisplay.h:
 * 
   //#define DISPLAY_WIDTH 128
   //#define DISPLAY_HEIGHT 64
   
   #define DISPLAY_WIDTH 96
   #define DISPLAY_HEIGHT 64
 *  
*/
```

```
/**
 * The MIT License (MIT)
 *
 * Copyright (c) 2016 by Daniel Eichhorn
 * https://github.com/ThingPulse/esp8266-oled-ssd1306/issues ;
 *
 */

/*
 * for OLED 64x48:
 * x-offset=32
 * y-offset=16
 * 
 * exchange  in OLEDDisplay.h:
 * 
   //#define DISPLAY_WIDTH 128
   //#define DISPLAY_HEIGHT 64
   
   #define DISPLAY_WIDTH 96
   #define DISPLAY_HEIGHT 64
 *  
*/

// Include the correct displayBRK library
// For a connection via I2C using Wire include
#include <Wire.h>  // Only needed for Arduino 1.6.5 and earlier
#include "SSD1306.h" // alias for `#include "SSD1306Wire.h"`

/*
// Include custom images
#include "images.h"
*/

// Initialize the OLED displayBRK using Wire library
SSD1306  displayBRK(0x3d, D2, D1);

// SSD1306  displayBRK(0x3c, D3, D5);
// SH1106 displayBRK(0x3c, D3, D5);

int counter = 1;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  Wire.begin();
  Wire.pins(D2,D1);

  // Initialising the UI will init the displayBRK too.
  displayBRK.init();

  displayBRK.flipScreenVertically();
  displayBRK.setFont(ArialMT_Plain_10);

}

void drawFontFaceDemo() {
    // Font Demo1
    // create more fonts at http://oleddisplayBRK.squix.ch/
    displayBRK.setTextAlignment(TEXT_ALIGN_LEFT);
    displayBRK.setFont(ArialMT_Plain_10);
    displayBRK.drawString(0+32, 0+16, "Hello world");
    displayBRK.setFont(ArialMT_Plain_16);
    displayBRK.drawString(0+32, 10+16, "Hello world");
    
    //displayBRK.setFont(ArialMT_Plain_24);
    //displayBRK.drawString(0, 26, "Hello world");
}

void drawTextFlowDemo() {
    displayBRK.setFont(ArialMT_Plain_10);
    displayBRK.setTextAlignment(TEXT_ALIGN_LEFT);
    displayBRK.drawStringMaxWidth(0+32, 0+16, 64,
    "Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore." );
}

void drawTextAlignmentDemo() {
    // Text alignment demo
  displayBRK.setFont(ArialMT_Plain_10);

  // The coordinates define the left starting point of the text
  displayBRK.setTextAlignment(TEXT_ALIGN_LEFT);
  displayBRK.drawString(0+32, 10, "Left aligned");

  // The coordinates define the center of the text
  displayBRK.setTextAlignment(TEXT_ALIGN_CENTER);
  displayBRK.drawString(32+32, 22, "Center aligned");

  // The coordinates define the right end of the text
  displayBRK.setTextAlignment(TEXT_ALIGN_RIGHT);
  displayBRK.drawString(64+32, 33, "Right aligned");

}

void drawRectDemo() {
      // Draw a pixel at given position
    for (int i = 0; i < 10; i++) {
      displayBRK.setPixel(i, i);
      displayBRK.setPixel(10 - i, i);
    }
    displayBRK.drawRect(12, 12, 10, 10);

    // Fill the rectangle
    displayBRK.fillRect(14, 14, 12, 12);

    // Draw a line horizontally
    displayBRK.drawHorizontalLine(0, 30, 20);

    // Draw a line horizontally
    displayBRK.drawVerticalLine(20, 0, 20);
}

void drawCircleDemo() {
  for (int i=1; i < 8; i++) {
    displayBRK.setColor(WHITE);
    displayBRK.drawCircle(16, 16, i*3);
    if (i % 2 == 0) {
      displayBRK.setColor(BLACK);
    }
    displayBRK.fillCircle(32, 32, 32 - i* 3);
  }
}

void drawProgressBarDemo() {
  int progress = (counter / 5) % 100;
  // draw the progress bar
  displayBRK.drawProgressBar(0, 16, 60, 10, progress);

  // draw the percentage as String
  displayBRK.setTextAlignment(TEXT_ALIGN_CENTER);
  displayBRK.drawString(64, 15, String(progress) + "%");
}

/*
void drawImageDemo() {
    // see http://blog.squix.org/2015/05/esp8266-nodemcu-how-to-create-xbm.html
    // on how to create xbm files
    displayBRK.drawXbm(34, 14, WiFi_Logo_width, WiFi_Logo_height, WiFi_Logo_bits);
}
*/

void loop() {
  //------------------------------------
  // clear the displayBRK
  displayBRK.clear();
  drawFontFaceDemo();
  
  displayBRK.setTextAlignment(TEXT_ALIGN_LEFT);
  displayBRK.drawString(0+32, 32+16, String(millis()));
  
  displayBRK.display();

  delay(1000);

  //------------------------------------
  displayBRK.clear();
  drawTextFlowDemo();
  
  displayBRK.display();

  delay(1000);

  //------------------------------------
  displayBRK.clear();
  drawTextAlignmentDemo();

  displayBRK.display();

  delay(1000);

 
  //------------------------------------ 
  counter++;
  delay(1000);
}
```

kaldoran, 2018-03-14:
Thanks for sharing your work @LimaLima, just out of information I will reuse your work in order to adapt all the code to all screen size ( using #define after including the library).

Then with this that should make 64 x 48 screen.
I will start this week-end and will update my comment as soon as a beta version comes out.

Testato, 2019-05-19:
This PR solve this Issue https://github.com/ThingPulse/esp8266-oled-ssd1306/pull/198
Please accept it

closed 2019-11-29 by stale[bot]

LennartHennigs, 2020-06-02:
please reopen this issue and apply PR