ThingPulse/esp8266-oled-ssd1306 #79: Cyrillic fonts

utya1988, 2016-08-14:
Hello. I really like you project WeatherClock, but i cannot use cyrillic fonts. and  i don't know to do this. Thanks.

FWeinb, 2016-08-25:
I don't know the inner workings of Cyrillic fonts. I worked on Arabic fonts in #56 but didn't have a solution at that time. In the future there will be the possibility to write your own custom mapper function for mapping Unicode to the internal character table.

FWeinb, 2017-01-05:
I think this is out of scope of this library. The building blocks will be in `3.2.4` but as none of us can speak cyrillic or arabic it is hard for us to implement this.

closed 2017-01-05 by FWeinb

colesnicov, 2019-08-03:
Hi.
Any progress in promoting cyrilic and extended fonts?

Voha888, 2019-11-24:
> peak cyrillic or arabic it is hard for us to implement this.

Thanks, cool solution, I'll try to use it. On your part, there is a proposal to consider the possibility of adding UTF-8 encoding support to the font files, not just ASCII, this will allow users to create multilingual fonts themselves