ThingPulse/esp8266-oled-ssd1306 #81: Dual displays

middelbh, 2016-08-25:
Hi! I though to be smart and connect two 0.96" oled displays to my EPS8266, each with the same address but each with different SDA and SDC ports (as I still have sufficient of them in the flight tracker..)

So this is part of the code:
.......
const int I2C_display_ADDRESS = 0x3c;
const int SDA_PIN = 4;
const int SDC_PIN = 2;
const int SDA_PIN2= 12;
const int SDC_PIN2= 13;
SSD1306Wire     display1(I2C_display_ADDRESS, SDA_PIN, SDC_PIN);
SSD1306Wire     display2(I2C_display_ADDRESS, SDA_PIN2, SDC_PIN2);**
......
Compiling the program however gave the following error:

.....
sketch\flightscan_test3.ino.cpp.o: In function `__static_initialization_and_destruction_0':
C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_472887/flightscan_test3.ino:195: undefined reference to `vtable for OLEDDisplay'
collect2.exe: error: ld returned 1 exit status
exit status 1
.....

My question: is it possible to have dual displays without changing the I2C addresses in this way?

FWeinb, 2016-08-25:
I just tested it using Arduino 1.6.9 with this code: 
```c++
#include <SSD1306Wire.h>

const int I2C_display_ADDRESS = 0x3c;
const int SDA_PIN = 4;
const int SDC_PIN = 2;
const int SDA_PIN2= 12;
const int SDC_PIN2= 13;

SSD1306Wire display1(I2C_display_ADDRESS, SDA_PIN, SDC_PIN);
SSD1306Wire display2(I2C_display_ADDRESS, SDA_PIN2, SDC_PIN2);

void setup() {
}

void loop() {
}
```

And everything is compiling, could you please post the complete sketch you are trying to run.

middelbh, 2016-08-25:
Hi,

 

Attached my sketch..

I added several extra’s, they work without problem. Just when I add the line

 

SSD1306Wire display2(I2C_display_ADDRESS, SDA_PIN2, SDC_PIN2);

 

The compilation gives errors…

 

I hope you can help!

 

Regards,

 

hans

FWeinb, 2016-08-25:
Sorry but the sketch isn't attached.

middelbh, 2016-08-25:
I tried to attach it as an rtf, but here the .ino…

FWeinb, 2016-08-25:
I looks like it isn't possible to attach it via email.

middelbh, 2016-08-25:
I will try this afternoon to attach ..

middelbh, 2016-08-25:
Her is the zipped .ino file.
Hope you can find the problem!

Hans
[flightscan_test4.zip](https://github.com/squix78/esp8266-oled-ssd1306/files/436951/flightscan_test4.zip)

middelbh, 2016-08-25:
By the way, in the .zip I only have the mentioned programlines, I do not yet use display2 furtheron in the sketch as the compile did not succeed.

FWeinb, 2016-08-27:
So had a minute to look a this. Kinda strange. I boiled it down to this issue: 

```
#include "SSD1306Wire.h"

const int I2C_display_ADDRESS = 0x3c;
const int SDA_PIN = 4;
const int SDC_PIN = 2;

SSD1306Wire     display1(I2C_display_ADDRESS, 4, 2);
SSD1306Wire     display2(I2C_display_ADDRESS, 5, 3);

//String thisWillBreakIt;

void setup(){}
void loop(){}
```

when you uncomment `String thisWillBreakIt` definition it will break the code. I have honestly no idea why this is happening.

closed 2016-08-27 by FWeinb

middelbh, 2016-08-27:
Thanks a lot! I will try ASAP! Interesting to understand why this happens!

Hans

middelbh, 2016-08-29:
Hi, I did as proposed, but same error again...

C:\Users\Hans\AppData\Local\Temp\arduino_modified_sketch_76867/flightscan_test4.ino:196: undefined reference to `vtable for OLEDDisplay'

collect2.exe: error: ld returned 1 exit status