ThingPulse/esp8266-oled-ssd1306 #82: Mangled display with SH1106 1.3" SPI OLED

fowlerk, 2016-08-27:
Hi, great work on this library, thanks much for your efforts.  I tried converting existing code that worked for an SPI SSD1306 .96" display to use an SH1106 1.3" (SPI) one.  Though I do get some recognizable display elements, the overall result is quite mangled.  Some random overwrites, some reverse display elements.  I know the code is new for the SH1106...has this been tested on the SPI version?

FWeinb, 2016-08-27:
There is currently no release with support for SPI but there is support for Sh1106SPI on the 'master' branch. Just clone this repo into the library folder and use `#include "SH1106Spi.h"` until we have time to do a release.

fowlerk, 2016-08-31:
Okay, yes, I realize SH1106 support is not yet "official", but the way you suggest is indeed what I had attempted.  Just relaying that even following this suggestion that the results are a mangled display.  For what it's worth, I'm using the weather station project as an example to test.

FWeinb, 2016-09-01:
When I have time I will look into this. What kind of display do you have? Does it use a CS pin, because that is currently not implemented.

fowlerk, 2016-09-01:
Yes, it is a 7-pin model (this one http://www.aliexpress.com/item/1PCS-1-3-inch-132X64-OLED-module-white-on-black-3-wire-SPI-4-wire-SPI/1472540455.html), which includes the CS line.  I believe I've used other SPI displays (likely with other libraries though) where the CS line was not connected.  I know this would be required where there are multiple devices on the same bus, but wasn't sure if it was necessary with a single device.

fowlerk, 2016-09-05:
Fabrice...I believe the issue was as a result of the CS line not being implemented as you mentioned.  After your comment on this, I looked closer at the examples and noticed this was not included in the calls to initialize the display routines for the SH1106.  Once I removed this from the initialization I was able to get this to work correctly.  Sorry for any confusion I may have caused.

closed 2016-09-05 by fowlerk