ThingPulse/esp8266-oled-ssd1306 #86: sprintf format

danbicks, 2016-10-03:
Hi Daniel, 

I must say firstly awesome work you have done, great examples.

I am trying to get the full digits of a clock to display, initial values are int's using sprintf to format so that leading zero is not truncated and displayed.  for example 07:01  and not 7 : 1

Any ideas on where I may be going wrong?

char bufHR[3]  = "";
sprintf(bufHR, "%2d", RTC_Hour); // format to 2 decimal places for display output
display.drawString(15, 20, bufHR);

Display works but still only displays single digit and not leading zero.

Thanks

Dans

hallard, 2016-10-03:
@danbicks 
`%02d` is the trick

```
sprintf(bufHR, "%02d", RTC_Hour); // format to 2 decimal places for display output
```

danbicks, 2016-10-03:
Thanks Buddy,

Already added Woo looks so cool.

Dans

squix78, 2016-10-03:
Cool. Now I learned something as well. I thought sprintf is not available for Arduino environment. Can we close this issue then?

danbicks, 2016-10-03:
Hi Squix,

Yep all closed, PS have you managed to get compass display code working?
Sneak download pretty please :)

Cheers buddy

Dans

closed 2017-01-05 by FWeinb