ThingPulse/esp8266-oled-ssd1306 #90: Add buffer scroll support

MichalKrakow, 2016-10-23:
In order to create something like console for esp it would be awesome to type only at bottom of screen and scroll content up to clear for new line.

FWeinb, 2016-10-23:
Have a look at [text logging](https://github.com/squix78/esp8266-oled-ssd1306/blob/master/UPGRADE-3.0.md#text-logging) this should be exactly what you are looking for. Please reopen this issue when you need something else.

closed 2016-10-23 by FWeinb

MichalKrakow, 2016-10-24:
Thanks for reply. 
It's good for some cases but form lib itself it would be awesome to get text warp. Maybe based on actual font chars width to get most of the display width.

FWeinb, 2016-10-24:
`c
// Draws a String with a maximum width at the given location.
// If the given String is wider than the specified width
// The text will be wrapped to the next line at a space or dash
void drawStringMaxWidth(int16_t x, int16_t y, int16_t maxLineWidth, String text);
`

Maybe this is what you are looking for?

tablatronix, 2017-01-03:
When is next milestone, when you going to cut a release for this ?