ThingPulse/esp8266-oled-ssd1306 #107: Support for 2 displays on the same espeasy device

jongerenchaos, 2017-01-14:
Thanks for this scrolling option it works OK with one display! When you add another second OLED display (3C & 3D) it doesn't work and one display displayed only the WIFI logo. Can you fix this so that everyone can use two of this displays?

FWeinb, 2017-01-15:
You you pleas add a sketch that shows this problem in action. Otherwise we can't really help you with your problem.

jongerenchaos, 2017-01-21:
For now i use the following build:
Build:	147
Core Version:	2_3_0

And i add 2 displays within the webconfiguration module (see all the uploaded pictures). When i restart the esp device only one of the two work. I try to made difference settings but none of all works.
I hope that this is enough information
[image]
[image]
[image]

closed 2019-12-04 by stale[bot]