ThingPulse/esp8266-oled-ssd1306 #109: Support for ESP8266 ESP-07

duongthuyvy1995, 2017-02-13:
Dear everyone,
I use ESP-07 and Oled spi but it doesn't work. I don't know this library support for ESP07 or not?
Thanks in advance!

squix78, 2017-02-13:
I think the ESP-07 does not expose the necessary pins for SPI. Better go for the I2C display, you'll need only 2 pins...

duongthuyvy1995, 2017-02-13:
Thanks @squix78
I found ESP-07 datasheet http://www.exp-tech.de/pdf/products/ESP-07/Ai-thinker%20ESP-07%20WIFI%20Module-EN.pdf 
SPI pins: IO12(MISO), IO13(MOSI), IO14(CLK), IO15(CS)
I connect: // Initialize the OLED display using SPI
#define D7   13 // D7 -> MOSI (DOUT)
#define D5   14 // D5 -> CLK
#define D2   4 // D2 -> DC
#define D8   15 // D8 -> CS
#define D0   16 // D0 -> RES

I uploaded code success, but Oled doesn't display anything. Connect pins like that right or wrong Sir?

closed 2019-12-04 by stale[bot]