ThingPulse/esp8266-oled-ssd1306 #110: Support for ESP32

arcao, 2017-02-14:
Hi,

I successfully tested this library with **SparkFun ESP32 Thing** and **SSD1306** driver via **I2C** and it works without any change. I also compiled example with SPI version of this driver without any problem, but I don't have SPI OLED display to test.

So you can add ESP32 platform/architecture to `library.json` and `library.properties`.

Jorgen-VikingGod, 2017-02-14:
I also tested it with I2C. Maybe on weekend I have some time left to test it on SPI too.

squix78, 2017-02-17:
Thank you guys! I updated the properties file and made a new release. Can you confirm that the library now shows up when using the ESP32 platform goal?

arcao, 2017-02-17:
Fix PIO platforms in `library.json`: https://github.com/squix78/esp8266-oled-ssd1306/pull/113

closed 2019-12-04 by stale[bot]