ThingPulse/esp8266-oled-ssd1306 #116: Is it possible to write " Æ, Ø, Å " on SSD1306?

Carlium, 2017-03-12:
I have tried to translate english text into my SSD1306, but there is one thing that annoys me:
I can't type "Æ, Ø or Å" (Danish and Norwegian letters) into the SSD1306, it's just blank. 
help please?

ltvanderkrogt, 2017-03-12:
If you can't find your font you can create Danish fonts with this? 
http://guy.carpenter.id.au/gaugette/blog/2012/11/11/font-support-for-ssd1306/

Carlium, 2017-03-13:
Thanks, but i'd rather have a guide on how to add a letter in the script file if possible. :-)

gerleimarci, 2017-08-07:
And please add Hungarian characters like ö, Ö, ü, Ü, ó, Ó, ő, Ő, ú, Ú, é, É, á, Á, ű, Ű, í, Í. But the best solution would be to add an option to upload .ttf files that are contains these special characters to the online converter.

closed 2019-12-04 by stale[bot]