ThingPulse/esp8266-oled-ssd1306 #118: Any simple example?

ceremcem, 2017-03-23:
Isn't there any "Hello World!" example? It would be quite useful to get the library up and running.

FWeinb, 2017-03-30:
You can find a simple demo [here](https://github.com/squix78/esp8266-oled-ssd1306/tree/master/examples/SSD1306SimpleDemo).

ceremcem, 2017-03-30:
Firstly, IMHO, it's quite impolite attitude that you just give an answer and then close an issue. Let me decide if your answer works  for me or not. If it works, I will use that button, no worries :) 

Secondly, [that demo](https://github.com/squix78/esp8266-oled-ssd1306/tree/9bc04a516ffa745e16e60e3402cc5cc8313c8cbd/examples/SSD1306SimpleDemo) is not simple, it's not the bare example. A "Hello World!" example requires the minimum lines of code. Is [including custom images](https://github.com/squix78/esp8266-oled-ssd1306/blob/9bc04a516ffa745e16e60e3402cc5cc8313c8cbd/examples/SSD1306SimpleDemo/SSD1306SimpleDemo.ino#L41) necessary? Obviously not. I will create a PR in a day or two that demonstrates my needs. 

Third: Please re-open this issue, as it's not resolved yet.

ltvanderkrogt, 2017-03-31:
Hi ceremcem, 
I made this bare minimum and it yells " hello world". But....my display was I2C. 
https://github.com/ltvanderkrogt/esp8266/blob/master/SSD1306SimpleDemoI2C.ino

closed 2019-12-04 by stale[bot]