ThingPulse/esp8266-oled-ssd1306 #125: Compiling error

Jakewd, 2017-05-31:
Im not sure where to put this but when running one of your examples I get this error.

Arduino: 1.6.8 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 4M (3M SPIFFS), nodemcu"

SSD1306SimpleDemo:61: error: 'D3' was not declared in this scope
 SSD1306  display(0x3c, D3, D5);
                        ^
SSD1306SimpleDemo:61: error: 'D5' was not declared in this scope
 SSD1306  display(0x3c, D3, D5);
                            ^
exit status 1
'D3' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Do you have any suggestions of what I should do?

~Jake

squix78, 2017-05-31:
Hi Jake

This usually means that you have not selected a NodeMCU or Wemos ESP8266 module in the Arduino Tools menu. Your message even says "Generic Module". If you are doing this with an NodeMCU or Wemos module then please select that board in the menu. If you are using another module then please replace D3 and D5 with the respective integer numbers or add #define D3 0 and #define D5 14 statements. You can see the mapping here: https://cdn.instructables.com/FPV/E4YC/IKLFP40J/FPVE4YCIKLFP40J.MEDIUM.jpg
Please close this issue if you agree that it has been resolved...

Kind regards,
Daniel

Jakewd, 2017-06-01:
Thank you for you help

closed 2017-06-01 by Jakewd