ThingPulse/esp8266-oled-ssd1306 #126: compile error (ESP8266)

tofrnr, 2017-06-04:
hallo,
ich kann den Beispiel Sketch nicht kompilieren 
 	SSD1306SimpleDemo.ino

Bibliothek esp8266-oled-ssd1306-master in Version 3.2.7 im Ordner: D:\Arduino\patch_libs\Arduino\libraries\esp8266-oled-ssd1306-master  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

ich habe auch den Verdacht, dss die I2C Pin Settings falsch sein könnten, denn das ESP8266 Board hat
I2C SCL: GPIO14  D5
I2C SDA: GPIO2   D4

- keinesfalls aber  wie im Sketch verwendet
SSD1306  display(0x3c, D3, D5);

allerdings funktionieren auch  nicht
SSD1306  display(0x3c, D4, D5);
SSD1306  display(0x3c, 2, 14);

beides dürfte aber zu keinem Compile Error führen, höchstens zu einem Laufzeitfehler (CMIIW).

squix78, 2017-06-04:
Hi tofrnr

Let's switch to English so more people can profit from this. Please turn on "Show verbose output during compilation" in your Arduino settings and put the last part of the output here. With the provided output is impossible to give a good help...

Cheers,
Dani

tofrnr, 2017-06-04:
this is my compete error message:

Arduino: 1.8.2 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (3M SPIFFS), nodemcu, Disabled, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\hannah\AppData\Local\Arduino15\packages -hardware D:\Arduino\patch_libs\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\hannah\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\patch_libs\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M3M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_151620 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\hannah\Desktop\BRK_IoT-Set_Examples\BRK_IoT-Set_Examples\Beispiel_6.4.1_squix\Beispiel_6.4.1_squix.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\hannah\AppData\Local\Arduino15\packages -hardware D:\Arduino\patch_libs\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\hannah\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries D:\Arduino\patch_libs\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=4M3M,ResetMethod=nodemcu,Debug=Disabled,DebugLevel=None____ -ide-version=10802 -build-path C:\Users\hannah\AppData\Local\Temp\arduino_build_151620 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\hannah\Desktop\BRK_IoT-Set_Examples\BRK_IoT-Set_Examples\Beispiel_6.4.1_squix\Beispiel_6.4.1_squix.ino
Using board 'generic' from platform in folder: C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_151620/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "C:\Users\hannah\AppData\Local\Temp\arduino_build_151620\sketch\Beispiel_6.4.1_squix.ino.cpp" -o "nul"
"C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\hannah\AppData\Local\Temp\arduino_build_151620/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\generic" "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire" "C:\Users\hannah\AppData\Local\Temp\arduino_build_151620\sketch\Beispiel_6.4.1_squix.ino.cpp" -o "nul"
"C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\har [... cut ...]
C:\Users\hannah\Desktop\BRK_IoT-Set_Examples\BRK_IoT-Set_Examples\Beispiel_6.4.1_squix\Beispiel_6.4.1_squix.ino:41:20: fatal error: images.h: No such file or directory

 #include "images.h"

                    ^

compilation terminated.

Bibliothek Wire in Version 1.0 im Ordner: C:\Users\hannah\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\Wire  wird verwendet
Bibliothek esp8266-oled-ssd1306-master in Version 3.2.7 im Ordner: D:\Arduino\patch_libs\Arduino\libraries\esp8266-oled-ssd1306-master  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

squix78, 2017-06-04:
looks like you are missing the images.h file: : fatal error: images.h: No
such file or directory

Which demo you are trying to compile? The UI Demo? when I open it through
the file > examples > SSD1306... > UiDemo then I get an images.h

Maybe you renamed something?

tofrnr, 2017-06-04:
1st time it was from the original folder, 2nd time from my working dir.
so again, from original folder:

exit status 1
'D3' was not declared in this scope

tofrnr, 2017-06-04:
changed that to
SSD1306  display(0x3c, 2, 14);

squix78, 2017-06-04:
You get that error, if you aren not using neither NodeMCU nor Wemos module.
Either replace the D* pins with an integer or add a #define D3 ... statement

squix78, 2017-06-04:
And did it work? Please close this issue if you agree that the issue has
been resolved...

tofrnr, 2017-06-04:
i don't know what a  NodeMCU nor Wemos module is or how to get it. I just installed the general ESP8266 board.
Strangely indeed the img file was not copied to the new sketch name to the new folder, but now itis in there.
D3, D5 is definetively wrong, but with
2, 14 (D4, D5) there is no compile error any more.

Anyway, the display output is broken, I see some flickering and random dot pixels and now an then a small part of a shape  at the bottom of the scrren, then vanishing quickly.
So now just the output is broken ftm.

tofrnr, 2017-06-04:
[image]

squix78, 2017-06-04:
Can you please post a picture of your hardware. It is really hard to help
without information...

tofrnr, 2017-06-04:
yes, you're welcome!
I have 2 identical ones, both working fine for my Raspberry Pi but both have identical issues for your lib unfortunately. 
t's not oroiginally from Adafruit but SSD1306 128x64 though.

[image]

tofrnr, 2017-06-04:
this is how it lloks for my Raspi:

[image]

tofrnr, 2017-06-04:
ps,
meanwhile also tried
#include "SH1106.h"
instead of
#include "SSD1306.h"

and now it works!
thank you very much,  issue can be closed!

closed 2017-06-04 by tofrnr

squix78, 2017-06-04:
Please also send one from your ESP module...

squix78, 2017-06-04:
Actually you might be using the wrong driver. This is a 1.3" display which usually has the SH1101 chip. Use this header instead and do proper instantiation as explained in the readme: https://github.com/squix78/esp8266-oled-ssd1306/blob/master/SH1106.h

tofrnr, 2017-06-04:
yes, although it was sold as a SSD1306 from a German vendor  it seems to be a 1.3" SH1106, and as a
SH1106 display(0x3c, D4, D5);
it eventually worked!