ThingPulse/esp8266-oled-ssd1306 #129: how to flip the display between normal and ve´flipvertical and back to normal?

tofrnr, 2017-06-08:
how is it possible to flip the display between normal and ve´flipvertical and back to normal?
sth like
```
if (digitalRead(D2)==LOW) { display.flipScreenVertically(); } // by a permanent jumper or switch
else
if (digitalRead(D2)==HIGH) { display.normalflip(); }
```

CelliesProjects, 2017-06-09:
I am running into the same issue here.

tofrnr, 2017-06-11:
that is no option, because when having D2=LOW then display.init will be called repeatedly or other way round for D2=HIGH display.flipScreenVertically() will be called repeatedly whixh woud lead to extremely flickering displays.

tofrnr, 2017-06-11:
no, does not work:

```
if (digitalRead(D1)==LOW && !OLED_FLIPPED) { display.flipScreenVertically(); OLED_FLIPPED = true; }
  else {display.init(); OLED_FLIPPED = false; }
```

leads to perpetual MCU reset

CelliesProjects, 2017-06-15:
Maybe this will work 
```
bool OLEDflipped = false;

setup(){
  display.init();
}

loop(){
  static bool PINstate = digitalRead(D1);
  if ( PINstate != OLEDflipped ){
    OLEDflipped = PINstate;
    OLEDflipped ? display.flipScreenVertically() : display.init();
  }
}
```

CelliesProjects, 2017-06-15:
But I agree a function to flip the display from the lib would be preferable.

tofrnr, 2017-06-16:
#doesn't work as well

RudyFiero, 2017-09-20:
The code below does the vertical flips. I am using a switch input here but on my target device I have an orientation sensor. (switch was easier to use for testing)

`
void switchOrientation() {

  switch5State = digitalRead(switch5);

  if (switch5State != Orientation)
  {
    Orientation = switch5State;
    digitalWrite(led15, Orientation);

    if (switch5State == 1)
    {
      //COMSCANDEC 0xC8    SEGREMAP   0xA1          // Normal display
      Wire.beginTransmission(I2C_DISPLAY_ADDRESS);
      Wire.write(0x80);
      Wire.write(0xC8);
      Wire.endTransmission();

      Wire.beginTransmission(I2C_DISPLAY_ADDRESS);
      Wire.write(0x80);
      Wire.write(0xA1);
      Wire.endTransmission();

      ui.nextFrame();
    }
    else
    {
      //Flipped
      //COMSCANINC 0xC0   SEGREMAP   0xA0
      Wire.beginTransmission(I2C_DISPLAY_ADDRESS);
      Wire.write(0x80);
      Wire.write(0xC0);
      Wire.endTransmission();

      Wire.beginTransmission(I2C_DISPLAY_ADDRESS);
      Wire.write(0x80);
      Wire.write(0xA0);
      Wire.endTransmission();

      ui.nextFrame();
    }
  }
}
`
This is a brute force method. I do  ui.nextFrame(); to refresh the buffer. The overlay does not get updated  until the next transition so it is a bit messy until the next. I haven't figured out how to force that. 

I would have like to determine the current frame and then force a re-write of it but I couldn't find what the current frame was. And it looks like there is logic to prevent re-writing to the current frame, but I'm not certain about that. 

The device I have is a small remote control that is battery operated. I wanted the display to reflect the orientation of the device. I wish the output was cleaner but it gets fixed up on the next transition. I might just accept that.

[image]

R0binBl00d, 2018-06-04:
Had the same problem and found something great :-)

http://nerdclub-uk.blogspot.com/2016/03/adding-flip-function-to-adafruits.html

so now I can flip the display anytime at runtime with a simple click of a button:
```
display.clear();

  if(digitalRead(Button) == 1)
  {
    flip = !flip;
  }
  
  display.drawString(0, 0, text);

  if(flip)
  {
    display.flip();
  }
  // write the buffer to the display
  display.display();
```
(/Users/myself/Documents/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display)
In the OLEDDisplay.h file I added the single line (just below flipScreenVertically):

`void flip();`

And in the OLEDDisplay.cpp file, added the function:
Since I am not using Adafruit Librarys, but ESP8266 Oled, I had to change the code a bit:

```
void OLEDDisplay::flip(void){ 
      // loop through every byte in the buffer and flip it vertically. 
      // Since the display draws 8 vertical bits (per byte) we can just flip each byte 
      // from MSB to LSB first 
      uint8_t t; 
      uint8_t p; 
      int byteCount=((DISPLAY_WIDTH*DISPLAY_HEIGHT/8)/2)-1;
      int bufferEnd = (DISPLAY_WIDTH*DISPLAY_HEIGHT/8)-1;
      for (uint16_t i=0; i<=byteCount; i++) { 
            p=0; 
            t=buffer[i]; 
            for(int j=0; j<=7; j++){ 
                  p=p << 1; 
                  if((t & 0x01)>0){ p=p|1;} 
                  t=t >> 1; 
            } 
      
            t=buffer[bufferEnd-i]; 
            buffer[bufferEnd-i]=p; 
      
            p=0; 
            for(int j=0; j<=7; j++){ 
                  p=p << 1; 
                  if((t & 0x01)>0){ p=p|1;} 
                  t=t >> 1; 
            } 
            buffer[i]=p; 
      } 
}
```

closed 2019-12-04 by stale[bot]