ThingPulse/esp8266-oled-ssd1306 #137: compilation error for board lolin32 + oled

twanek, 2017-07-13:
hi!

i try to use this board: https://www.aliexpress.com/item/Wemos-18650-Battery-shield-V3-RaspberryPi-Arduino-Lolin-ESP32-OLED-wemos-for-Arduino-ESP32S-WiFi-Modules/32807898232.html

with this lib, but i've got a compile error:

```
In file included from C:\ARDUINO\libraries\esp8266-oled-ssd1306/SSD1306Wire.h:31:0,

                 from C:\ARDUINO\libraries\esp8266-oled-ssd1306/SSD1306.h:30,

                 from C:\Users\wanek\AppData\Local\Temp\arduino_modified_sketch_891573\SSD1306SimpleDemo.ino:29:

C:\ARDUINO\libraries\esp8266-oled-ssd1306/OLEDDisplay.h: In member function 'virtual bool OLEDDisplay::connect()':

C:\ARDUINO\libraries\esp8266-oled-ssd1306/OLEDDisplay.h:253:29: error: no return statement in function returning non-void [-Werror=return-type]

     virtual bool connect() {};

                             ^

In file included from C:\ARDUINO\libraries\esp8266-oled-ssd1306/SSD1306.h:30:0,

                 from C:\Users\wanek\AppData\Local\Temp\arduino_modified_sketch_891573\SSD1306SimpleDemo.ino:29:

C:\ARDUINO\libraries\esp8266-oled-ssd1306/SSD1306Wire.h: In member function 'virtual void SSD1306Wire::display()':

C:\ARDUINO\libraries\esp8266-oled-ssd1306/SSD1306Wire.h:83:23: warning: comparison is always false due to limited range of data type [-Wtype-limits]

         if (minBoundY == ~0) return;

                       ^

cc1plus.exe: some warnings being treated as errors

Using library Wire at version 1.0 in folder: C:\ARDUINO\hardware\espressif\esp32\libraries\Wire 
Using library esp8266-oled-ssd1306 at version 3.2.7 in folder: C:\ARDUINO\libraries\esp8266-oled-ssd1306 
exit status 1
Error compiling for board WEMOS LOLIN32.
```

the only thing i modified in the SSD1306SimpleDemo example is at line 61 replaced D3, D5 with 5, 4 as per seller instruction.

thanks!

twanek, 2017-07-13:
ok, after some testing, i figured out that the example code on the aliexpress site was given for the esp8266 version of this board, not esp32. (if i set in arduino ide board type esp8266, it compiles fine, but if i set esp32 it gives the compilation error)

sp, the question is, what should i use instead of D3 and D5 with the esp32 board? (for i2c)

schumar, 2017-07-23:
The pin numbers given by the vendor are correct.

I guess the issue is a newer version of gcc, which is a bit more strict regarding "clean" code.
Please try File -> Preferences -> Compiler warnings: Default

twanek, 2017-07-23:
thanks for the advice, it compiles now!
(i do not have access right now to the hw to upload / test, but i think should be ok.

the solution is interesting for me, because i never thought that the compiler warnings in the preferences are actually modify the compilation process...  i thought that it influences only the message output in the window.

beosro, 2017-11-26:
Hi. Well, I have the same problem with ESP32 and Oled (Heltech board). Same warning and error. The solution with "Default" for the compiler warnings gives the same result, error and stops the process.
On a mac mini with El Capitan the same code compiles fine. Here, on another mac mini with Sierra ... bof.
Any clues?
Thanks,
Alex

closed 2019-12-04 by stale[bot]