ThingPulse/esp8266-oled-ssd1306 #140: Library compilation errors

G6EJD, 2017-08-17:
I get:
esp8266-oled-ssd1306-master/OLEDDisplay.h:253:29: error: no return statement in function returning non-void [-Werror=return-type]
     virtual bool connect() {};

SH1106Wire.h:90:23: warning: comparison is always false due to limited range of data type [-Wtype-limits]
         if (minBoundY == ~0) return;
#############
The first needs a return statement adding.
The second I think is typo and should be '!=' as I can find no reference to the '~=' operator.

vshymanskyy, 2017-12-04:
Confirm. This is reproducible only if Arduino IDE config "Compiler Warnings" is set to More or All

dsptech, 2018-01-07:
Same issue here. 
minBoundY is a uint8_t ` uint8_t minBoundY = ~0`,  meanwhile "~0" is assumed as int instead.
The right way to perform the check is: `if (minBoundY == (uint8_t) ~0) return;`

lorol, 2018-01-07:
Maybe it is good to replace all 18 hits in library' *.h files from "= ~0"   to    "= (uint8_t) ~0" since they always refer the 8bit unsigned minBoundY or minBoundX

dsptech, 2018-01-07:
A such typecast is already done in every assignment so there is no need to apply changes here, but every check  `minBoundY == ~0` (6 hits ?) will fail in it's purpose (even if only a warning is generated) due the integral promotion. 
In the checks the typecast is needed.
I think that a more polite way is to change  every occurrence of "~0" with UCHAR_MAX (or similar define), so, every integral promotion issue will be avoided.

closed 2018-07-11 by G6EJD