ThingPulse/esp8266-oled-ssd1306 #141: Support for Arduino / AVR ATmega CPU ?

kratochviljan, 2017-08-17:
Hello, first thank you for excellent OLED library. I'm using it on ESP8266 a lot!
Now I'd like to use it on generic Arduino AVR Atmega MCU but I can't.
I made change to architectures=* in library.properties file, but there are many errors during compilation. Hmmm, apparently these settings matters.
So my question is: do you have plan to add support for AVR Atmega MCU?

Todelec, 2017-12-21:
Hi,

changes I made to run library on Mega2650 with SH1106 display controller:

library.properties:
Add "avr" to architectures.
Should look like this `architectures=esp8266,esp32,avr`

SH1106Wire.h
...
`#ifdef __AVR`
`#define _min(x,y) min(x,y)`
`#define _max(x,y) max(x,y)`
`#endif`

`bool connect() {`
`   #ifdef __AVR`
`     Wire.begin();`
`   #else`
`      Wire.begin(this->_sda, this->_scl);`
`   #endif`
...

harlock974, 2019-08-01:
Hi
I tried Todelec fix for an Arduino UNO (ATMega 328), and change the déclaration to : 
`SH1106 display(0x3c, A4, A5);`
to take into account SDA and SCL pins of the UNO.

No errors in compilation, but the display remains blank.
Any ideas ?

closed 2020-02-11 by stale[bot]