ThingPulse/esp8266-oled-ssd1306 #145: How to get active frame number with Ui Library  -  QESTION

RudyFiero, 2017-09-20:
I'm trying to implement a user interface using the Ui Library. The problem is that I can't find anywhere to get the active frame being displayed. Without knowing the frame I can't match the user input with anything. 

How can I make this UI a user interface?

RudyFiero, 2017-09-21:
I feel like such a hacker. 

I got it. But I'm not sure if there was a way to get this from the existing library without modification. If there is then please let me know. I say I feel like a hacker because I figured out how to do it but I don't think I have an understanding of what I did. In a general way yes, but it was mostly looking at something similar and trying to work with it. And fail. And fail again. Read more on the internet, and fail again. Take a break and then, fail again. Then I took a few steps back mentally and tried to understand better what was there. And tried again. And it worked. 

In OLEDDisplayUi.h I included this line.

`
 int8_t currFrame();  
 `

Then in OLEDDisplayUi.cpp I included,

```
int8_t OLEDDisplayUi::currFrame(){
  return this->state.currentFrame;
}
```
So simple. (compared to other thing I tried)

I had previously had one extra 'this->' and it didn't work. I really don't have a firm grasp of 'this'. 
So now that I have a way to get the frame number I can proceed with the rest of the user input logic.  I am going to leave this issue open just in case you think it is a good idea to implement the ability to determine the frame number. Or indicate how it could be done with the library as it is if it can be done, even though I couldn't see a way.

kjdolan99, 2018-04-05:
You can get the current frame without modifying the library by using `ui.getUiState()->currentFrame;`.
 `getUiState()' returns a pointer to the state structure, which represents the current state of the user interface. The arrow operator then accesses the 'currentFrame' member of the state structure.

closed 2018-04-21 by RudyFiero

guarrod, 2019-11-26:
Please someone can help me? i can't realize how to print this currentFrame on the serial port