ThingPulse/esp8266-oled-ssd1306 #147: SSD1306Wire: Option to use other speeds than 700kHz

Sthing, 2017-09-29:
Hi Daniel.

Thank you for your great library.

I have both an OLED and a PCF8574 I/O expander on the same I2C-bus.

My problem is that the PCF8574 is a 100kHz device, and SSD1306Wire::connect() sets the bus-speed to 700kHz.

I think it should be possible for the user of the library to control the clock frequency.

One possible solution is to provide an extra constructor: ```SSD1306Wire(uint8_t _address, TwoWire* wire)```. Then I can pass ```&Wire``` after setting it up like I want.

Another soultion is to add the wanted clock frequency to the existing constructor, and let 700kHz be the default: ```SSD1306Wire(uint8_t _address, uint8_t _sda, uint8_t _scl, unsigned int freq = 700000)```

I will gladly provide a pull request, but I would like to get your input first on which solution you prefer.

Best regards,
Søren Thing, Denmark

pikolo84, 2017-11-28:
any progress on this?

Sthing, 2017-11-28:
I am still waiting for an indication regarding Daniel Eichhorn preferred solution.

llinke, 2018-01-22:
Found this while facing the same problem, but seems to me, there is no issue or need to change anything.

Sure, `SSD1306Wire::connect()` does call `Wire.setClock(700000);`, but **Wire** is a static instance of Arduino/ESP8266's I2C core implementation, so you can easily call `Wire.setClock(100000);` or whatever at any time.

closed 2019-05-19 by marcelstoer