ThingPulse/esp8266-oled-ssd1306 #148: WebKit8 OLED display not working

palsbo, 2017-09-30:
I have an ESP8266 with buildin OLED (https://www.aliexpress.com/item/ESP8266-WIFI-Chip-0-91-inch-OLED-CP2014-32Mb-Flash-ESP-8266-Module-Internet-of-things/32822165059.html)
It useses SDA==pin 2 and SCL=pin 14, and it works with Adafruuit_SSD1306 library but only when I define by using `Adafruit_SSD1306 display(OLED_RESET);` 
where OLED_RESET = 4
"Normal OLED modules" require OLED = 0;
I gues this is the reason it is not working with your library.
Any ideas on how to get it to work?

zdar, 2017-10-15:
You have to use correct pins for constructor (supplied documentation is not correct):

SSD1306xxx display(0x3c, 4, 5);

and You have to reset display manually before library init

  // reset display
  pinMode(16, OUTPUT);
  digitalWrite(16, LOW); 
  delay(50);
  digitalWrite(16, HIGH);

spblinux, 2019-01-26:
@squix78: Would you mind to put the hint of zdar to the [readme page](https://github.com/ThingPulse/esp8266-oled-ssd1306#i2c-with-wireh) of your library?

It took me quite a while of reading to find this information and I can confirm that displays like TTGO ESP8266 0.91 inch OLED for arduino for nodemcu require this reset command; SDA=D4, SCL=D5, RST=D2 with 10ms delay in my case. (Luxury version would be to add it to the constructor.) - Used for [JLInfoTicker](https://github.com/androdlang/InfoTicker) which is based on ThingPulse ESP8266 OLED SSD1306.

closed 2019-08-08 by stale[bot]