ThingPulse/esp8266-oled-ssd1306 #150: Manually calling Frames

DuncanAmos, 2017-10-10:
Hi
I'm not a programmer type person, so I struggle a bit (more than a bit!!!)

I want to manually call frames, based on the return from a web server - the called frame to replace the previous one without any obvious sliding transition (i.e. so it appears instant).

I can use "ui.setTimePerTransition(1);" to get a visually instant transition between frames in my code based on the SSD1306UiDemo code, but I can't workout how to manually call individual frames. I tried various things, but nothing seemed to work.

Can somebody point me in the right direction - in simple terms?

Thanks

RudyFiero, 2017-10-10:
I'm not a programmer. And I have struggled with this too. But I think I have an answer for you.

disableAutoTransition();
switchToFrame(frame#);

/* Switch without transition to frame `frame`. */
void switchToFrame(uint8_t frame);

DuncanAmos, 2017-10-11:
Hi Rudy

Thanks for that.

It took me a while to discover that I needed to use "**ui.**disableAutoTransition();" and "**ui.**switchToFrame(frame#);", and that "frame#" is one less than the frame's title number (because it's in an array and arrays start from zero) but that does seem to work - sort of...

As I'm understanding it (or not!), AutoTransition has to be enabled first to achieve the first frame and then, at the end of that frame, I include "ui.disableAutoTransition();".

However, if I want to add a delay for the display of that first frame and then switch to another (as a 'splashscreen'), if I use this at the end of the first frame (0 in the array):

  ui.disableAutoTransition();
  delay(2000);
  ui.switchToFrame(1);

it ignores the delay completely and goes directly to Frame(1)...

It would be much simpler if I could dispense with the autotransitions, the callbacks (which I can't understand at all) and the fancy things, and just call individual frames in the sketch as and when they're needed in response to what's received from the server (just as I would call any other function under 'normal circumstances') but I can't work out how to do that.

DuncanAmos, 2017-10-13:
Hi Rudy

I've now made a lot of progress and have (kind of) worked out how to do what I need, but there is one problem that's still confusing me.

When I call a frame, there is a delay of around a second before the change takes place. This appears to relate to the library, rather than the return from the server, as it also does it if I call the frame directly from button press detection on the client.

For some of the frames, the delay wouldn't be particularly significant, but I have a countdown sequence that must remain synchronised (at least visually) with what's happening at the server.

If I add in some serial debug, then the call for the frame change is printed in the Serial Monitor and then, when the frame change actually occurs, a new blank line is printed, although I'm actually calling for that anywhere.

Any ideas???

Thanks

RudyFiero, 2017-10-13:
I will try an look at it tonight. It should be able to change without delay. I forget why but I had to call the next frame function for some reason and that allowed me to something right away. 

As far as a countdown timer. It seems to me that an element in the frame can be modified rather than redrawing the whole frame. The default example's clock makes me think that. I could be very wrong in that but if the clock can be updated without problems then I think other non-graphic data can as well.

DuncanAmos, 2017-10-13:
Thanks - it's got me very confused.

For the countdown timer, I did try and work out if it would be possible to just change the numbers. The Clock Example makes use of an overlay for the face of the clock and then draws the hands. I already have an overlay for the connected status (which also seems to suffer from a delay problem, but that may be my coding) and I wasn't sure if I could have more than one overlay at the same time.

Basically, I got lost trying to understand the code.

As the countdown is coming from the server (AP) it may make little difference whether I redraw the numbers or call a new frame, I suspect. The way I'm thinking at the moment is that it will receive "count5", "count4", etc. and then call the appropriate frame.

RudyFiero, 2017-10-13:
The demo I am using WeatherStationDemo.ino  as a starting point for my application. It now has five frames. The first one is the time. This is just large text, no drawn clock. The second frame is the current weather, the third frame is the forecast and this one has caused me problems. It has three graphic elements. 

Problem frame -  https://1.bp.blogspot.com/-O0G-gWU3JxY/Vlr40FI48kI/AAAAAAAAAdw/ROmQ3IoAx88/s1600/WeatherStationV2.JPG

In the main loop of my sketch I check remainingTimeBudget and if it is below 3 I don't do my own code. And during the forecast frame I have some significant response issues because my routines are not getting called. I expect to not have time during transitions and that is okay with me. I don't want input accepted as I want the input to be mated with the specific frame. 

The changing clock I mentioned before was purely text, and I have no issues when it is changing. The graphic clock could be doing a lot of processing and that I expect could cause you response delays. My device is battery operated so I am trying to keep the power consumption low. I did try using 160MHz for a bit when I was trying to figure out where my delays were coming from. I had been worried that my diagnostic print outputs might also be eating up my available working time.

If your countdown requires a new connection for each send then that might be too much overhead. It depends on how you have implemented that part of it. I mention it but I assume it is not the problem.

DuncanAmos, 2017-10-13:
> AAAAAAAAAdw/ROmQ3IoAx88/s1600/WeatherStationV2.JPG

I built the original weather station when Daniel Eichorn published it as an
Instructable (Weather Widget) a couple of years ago, using an ESP-01.

That works fine except that, for some reason, it keeps locking up (stuck on
one frame, clock not running) and requires a reboot.

That, of course, slides from frame to frame and that action would mask the
delay problem I'm seeing when I try to switch directly between them.

I think what's happening is that the frame appears, but there's a delay
before it signals that it's finished, allowing the next action in the
sketch to occur - but I may be wrong. I based that idea on 'gut feeling'
rather than anything else.

If you run the standard demo, there is a default frame duration somewhere
which controls that and it also relates to the default transition time (I
think).

Those should have no effect if the AutoTransitions are disabled but I
suspect they're not being eliminated/bypassed.

I've tried "ui.setTimePerFrame(1);" and "ui.setTimePerTransition(1);" but
they didn't seem to have any effect.

I don't understand the whole "TimeBudget" thing. If I put my code inside
the 'if statement', it won't work. If I remove the 'if statement' but keep
the first line "int remainingTimeBudget = ui.update();" it runs OK. If I
try and remove that first line, nothing happens with the display.

It could be possible, somehow, that this relates to the delay problem I'm
seeing.

RudyFiero, 2017-10-14:
I'm remembering some adjustments I made to the defaults. 
I set   ui.setTargetFPS(30);

It takes a certain time to draw & update a frame. Less frames per second means that there will be more time between frames to do other work. So try lower numbers but fast enough that it still looks okay. 

setTimePerTransition(200);

With a lower frame rate I think the transitions looked too jerky. (memory might not be accurate) But whatever the reason I thought that a faster transition with a lower number of frames would look better. The frames zip by a little faster and our brains merge the movement rather than notice the stops. In the end the fewer frames per second gives us more time to do other work. 

I got home late so I don't have much time to look at this, and I unexpectedly have to work tomorrow so I can't be up late. I'll let you know if I find anything useful.

DuncanAmos, 2017-10-14:
Hi Rudy

The FPS is the refresh rate - with static frames, that isn't really relevant. Similarly, the TimePerTransition is the time a new frame takes to 'slide in', so with switching directly from one frame to another, there is no actual transition time and that one isn't relevant either.

I'm tempted to dispense with the UI and call the frames directly, as is done in the 'SSD1306SimpleDemo' sketch, but as I want to have the current connection status displayed on every frame, and updating in real time, the Overlay option provided by the UI is the easiest method.

There are other ways of achieving the connection status display but they're all much less convenient.

It would be much better if I could find out what's causing the delay and remove the problem, but I just can't understand enough of the code...

RudyFiero, 2017-10-14:
> I'm tempted to dispense with the UI and call the frames directly

That was my thought. I just hoped the UI would work so you didn't have to go through implementing the parts you needed.

> The FPS is the refresh rate - with static frames, that isn't really relevant. 

I thought that if it was busy with a transition that it wouldn't allow you time to do your own processing. But while the frame is held in place it should not be using a lot of CPU time. I do think (not know) that the FPS is refreshing the image, like with the clock updates. But it doesn't have to do much as far as calculations. The reason I think the display is still being updated is that the forecast frame (with the three graphics) consumes a lot of CPU time even though nothing is being changed when that frame is held.

DuncanAmos, 2017-10-14:
Hi Rudy

What do you make of this:

If I copy the complete working sketch (CTRL+A, CTRL+C) and put it into a new sketch (CTRL+V) and then hit Compile, it throws this error:-

fatal error: images.h: No such file or directory

Nothing has been changed, it's exactly the same file but with a new name - how can the images.h file have disappeared?

It's obviously one of those files that's buried inside something else. or somewhere obscure, because I can't find it - but it must be there somewhere, or the original sketch wouldn't recompile quite happily.

I'm very confused...

RudyFiero, 2017-10-15:
In my sketch the image file is in a different tab in the Arduino IDE. Check to see if images.h is in the main file as an include.

I was too burnt out after work today so I haven't looked at this. Maybe tomorrow.

DuncanAmos, 2017-10-15:
Hi Rudy

You were quite correct. The images.h was in another tab - I just hadn't
noticed it.

The redesigned version, calling frames directly, is now working in basic
form - but the delay problem is still there, so it's not related to the UI
animation system I don't think.

I've put Serial.println() statements all over the place and I'm now
thinking that the delay may be due to the server not releasing the client
connection immediately after it's sent the data back, so the
'while(client.available) loop doesn't exit when I expected it to.

That's nothing to do with this library, so I'll mark this issue as closed -
but if you have any ideas, I'd be glad to hear them!

Thanks for your help...

Duncan

RudyFiero, 2017-10-15:
I'm glad you are moving forward on this. 
Good luck.

DuncanAmos, 2017-10-17:
Hi Rudy

I was a little premature in thinking I'd finished with the library's
eccentricities and my incompetence...

Having rewritten it to lose the arrays and callbacks and call frames
individually, I could get everything before the loop to work, or everything
in the loop - but not both...

Two days later I found it and fixed the delay problem - missing "()" and a
comma respectively.

Now to try and rationalise the code.

Thanks again.

RudyFiero, 2017-10-18:
>  Two days later I found it and fixed the delay problem - missing "()" and a
> comma respectively.

I hate mistakes like that. But at least you solved it.

closed 2019-12-04 by stale[bot]