ThingPulse/esp8266-oled-ssd1306 #152: display.init() return value

Gerry33, 2017-10-20:
Hi all,
first a big 'thumbs up' for this nice lib. 

Short question: display.init() returns true | false. But even when nothing is attached to the ESP, means 
no display at all is present, it returns 'true'. 

Now how to interpret if the display is initialised correctly ? 

Thanks for any comment 
Gerry

RestiveX, 2017-12-03:
It returns false only if there is not enough memory for the buffer!
If you want to check the connection with the I2C display - use this function:

`boolean find_i2c(uint8_t address) { Wire.beginTransmission(address); return (Wire.endTransmission () == 0); }`

`#define LCD_address    0x3c  // I2C address
display_enable = find_i2c(LCD_address);`

Gerry33, 2017-12-04:
Thank you very much !  Highly appreciated. 

Will implement asap. 

br 
gerry

closed 2017-12-04 by Gerry33