ThingPulse/esp8266-oled-ssd1306 #156: compilation error 'D2' was not declared in this scope

b3gi, 2017-11-19:
Hey. I ran into a problem. The program does not allow me to compile my project. An error: 'D2' was not declared in this scope.

#include <brzo_i2c.h>

#include <OLEDDisplay.h>
#include <OLEDDisplayFonts.h>
#include <OLEDDisplayUi.h>
#include <SH1106.h>
#include <SH1106Brzo.h>
#include <SH1106Spi.h>
#include <SH1106Wire.h>
#include <SSD1306.h>
#include <SSD1306Brzo.h>
#include <SSD1306Spi.h>
#include <SSD1306Wire.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>

#include <dummy.h>

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#define USE_DISPLAY /* <-- uncomment that if you wanna use the display */

#include <Wire.h>

//include the library you need
#include "SSD1306.h"
//#include "SH1106.h"

//button pins
#define upBtn D6
#define downBtn D7
#define selectBtn D5

#define buttonDelay 180 //delay in ms

//render settings
#define fontSize 8
#define rowsPerSite 8

//create display(Adr, SDA-pin, SCL-pin)
SSD1306 display(0x3c, D2, D1);
//SH1106 display(0x3c, D2, D1);

Error: 
esp8266_deauther:61: error: 'D2' was not declared in this scope

esp8266_deauther:61: error: 'D1' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino: In function 'void setup()':

esp8266_deauther:49: error: 'D6' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:377:11: note: in expansion of macro 'upBtn'

esp8266_deauther:50: error: 'D7' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:378:11: note: in expansion of macro 'downBtn'

esp8266_deauther:51: error: 'D5' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:379:11: note: in expansion of macro 'selectBtn'

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino: In function 'void loop()':

esp8266_deauther:49: error: 'D6' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:484:19: note: in expansion of macro 'upBtn'

esp8266_deauther:50: error: 'D7' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:492:26: note: in expansion of macro 'downBtn'

esp8266_deauther:51: error: 'D5' was not declared in this scope

C:\Users\Р“СЂРёРіРѕСЂРёР№\Desktop\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:500:26: note: in expansion of macro 'selectBtn'

In programming, I'm not strong and generally I'm R-RUS, so if there are grammatical errors, forgive me)
BYE BYE)

reivaxy, 2017-11-19:
Hey,

Try moving #include <Arduino.h> before the other include instructions.

reivaxy, 2017-11-21:
@b3gi any news ?

per1234, 2018-02-08:
The cause of this error is that the D*n* pin names are only defined for a subset of ESP32 and ESP8266 boards:
### ESP32
- IntoRobot Fig
- Microduino-CoreESP32
- u-blox NINA-W10 series (ESP32)
- Widora AIR

### ESP8266
- WeMos D1 R2 & mini
- WeMos D1 mini Pro
- Wemos D1 mini Lite
- WeMos D1 R1
- NodeMCU 0.9 (ESP-12 Module)
- NodeMCU 1.0 (ESP-12E Module)
- WifInfo

Compiling the examples with any other board selected will fail.

FrankenApps, 2018-08-16:
There is an easy fix though:
For ESP32 boards simply replace `SSD1306 display(0x3c, D2, D1);` with `SSD1306 display(0x3c, 21, 22);` and change the wiring accordingly...

closed 2019-12-04 by stale[bot]

georgedrive, 2019-12-17:
WeMos D1 R2:

#include <OLED_I2C.h>
//OLED  myOLED(D2, D1); 
OLED  myOLED(4, 5);

work