ThingPulse/esp8266-oled-ssd1306 #158: Wifi Kit 8 Simple Demo Text Compressed (Squished)

davidbitton, 2017-11-21:
I am running the Simple Demo example. Seems like an aspect ratio issue. Is there somewhere I can fix that? Thanks!

[image]
[image]
[image]

squix78, 2017-11-21:
Well, I don't know this hardware. What resolution does this display have?

davidbitton, 2017-11-21:
@squix78,
  128 x 32. I tried editing `DISPLAY_HEIGHT` in OLEDDisplay.h but that didn't work. It just wrote to half the screen. i attached the spec doc for the display. thanks.

[0.91 specification.pdf](https://github.com/squix78/esp8266-oled-ssd1306/files/1492846/0.91.specification.pdf)

AlbrechtMe, 2017-12-18:
Look for similar Boards on eBay - there is always the driver link:
"https://www.adrive.com/public/drzr3P/datasheet%20F000905.rar" with some data (pin diagram) .

A good review with a working example "WIFI-Kit-8-Test2.zip" (inclusive oled libraray) is there: 
[https://www.mikrocontroller.net/topic/438027](url).

The graphics routines are simple. I made a little routine to scroll up and print on the bottom line:
`void PrintKit8(String line)
{ display.scroll_up();
  display.draw_string(0,24,line.c_str());
  display.display();
}`

My board has 4M, I use it always with OTA. If programmed by serial, don't forget to set Reset Method to nodemcu.

masalinas, 2018-04-26:
@davidbitton I have the same hardware like you from (Heltec) I used the u8g2 but when I test the SSD1306SimpleDemo from esp8266-oled-ssd1306 I not see anything in the screen, I must configure something from the demo sketch??

davidbitton, 2018-04-26:
I'm not in front of the code but I think the OLED is not on the usual port; need to change it.

masalinas, 2018-04-26:
The example set this: SSD1306  display(0x3c, D3, D5); do you know what port I must to set for this device?
Regards

AlbrechtMe, 2018-04-29:
I don't use the fancy D3, D5 notation and prefere real GPIO numbers...
Better were `SSD1306Wire  display(0x3c, 4, 5, GEOMETRY_128_32); // addr, sda, scl, geometry`.
But I could not get it to work - I think, one has to name and use the RESET pin of the display.
In the given 'drivers' (look my  last post) there is a pinout with marked display ports.
The working example there (WIFI-Kit-8-Test2.ino) declares the display variable by:
`OLED display=OLED(4,5,16); //  sda, scl, reset`
Omitting the reset pin (16) there leaves also a blank display. For your conveniance you'll find my example with OTA here:
[Example_and_corrected_pinout.zip](https://github.com/ThingPulse/esp8266-oled-ssd1306/files/1958829/Example_and_corrected_pinout.zip)

davidbitton, 2018-04-30:
using `OLED.h` I have

```c
#include "OLED.h"

#define RST_OLED 16
OLED display(4, 5);
``` 
and in `setup()` I have
```c
  pinMode(RST_OLED, OUTPUT);
  digitalWrite(RST_OLED, LOW);   // turn D2 low to reset OLED
  delay(50);
  digitalWrite(RST_OLED, HIGH);    // while OLED is running, must set D2 in high
```

sschepis2, 2019-02-12:
Get this working? I had to call Wire.setClock(700000) to get the onboard Heltec display to work

anygenerator, 2019-07-18:
> Get this working? I had to call Wire.setClock(700000) to get the onboard Heltec display to work

Would you provide and example of the sketch you were able to get the screen working?

davidbitton, 2019-07-18:
```
#include <ESP8266WiFi.h>
#include <Arduino.h>
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <ClickEncoder.h>
#include <menu.h>
#include <menuIO/clickEncoderIn.h>
#include <menuIO/keyIn.h>
#include <menuIO/chainStream.h>
#include <menuIO/serialOut.h>
#include <menuIO/adafruitGfxOut.h>
#include "Logo.h"

using namespace Menu;

#define OLED_RESET 16

Adafruit_SSD1306 gfx(OLED_RESET);

const char* constMEM hexDigit MEMMODE="0123456789ABCDEF";
const char* constMEM hexNr[] MEMMODE={"0","x",hexDigit,hexDigit};
char buf1[]="0x11";//<-- menu will edit this text

int chooseTest=-1;
CHOOSE(chooseTest,chooseMenu,"Choose",doNothing,noEvent,noStyle
  ,VALUE("First",1,doNothing,noEvent)
  ,VALUE("Second",2,doNothing,noEvent)
  ,VALUE("Third",3,doNothing,noEvent)
  ,VALUE("Last",-1,doNothing,noEvent)
);
 
MENU(mainMenu,"Main menu",doNothing,noEvent,wrapStyle
  ,OP("Op1",doNothing,noEvent)
  ,EDIT("Hex",buf1,hexNr,doNothing,noEvent,noStyle)
  ,SUBMENU(chooseMenu)
  ,EXIT("<Back")
);

const colorDef<uint16_t> colors[] MEMMODE={
  {{WHITE,BLACK},{WHITE,BLACK,BLACK}},//bgColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//fgColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//valColor
  {{BLACK,WHITE},{BLACK,WHITE,WHITE}},//unitColor
  {{BLACK,WHITE},{WHITE,WHITE,WHITE}},//cursorColor
  {{BLACK,WHITE},{WHITE,BLACK,BLACK}},//titleColor
};

#define gfxWidth 84
#define gfxHeight 48
#define fontX 6
//5
#define fontY 9
#define MAX_DEPTH 2

// rotary encoder pins
#define encA    D8
#define encB    D6
#define encBtn  D7

ClickEncoder encoder = ClickEncoder(encA, encB, encBtn, 4);//simple quad encoder driver
ClickEncoderStream encStream(encoder, 1);// simple quad encoder fake Stream

MENU_INPUTS(in,&encStream);

MENU_OUTPUTS(out,MAX_DEPTH
  ,ADAGFX_OUT(gfx,colors,fontX,fontY,{0,0,gfxWidth/fontX,gfxHeight/fontY})
  ,SERIAL_OUT(Serial)
);

NAVROOT(nav,mainMenu,MAX_DEPTH,in,out);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  //
  // Set WiFi to station mode and disconnect from an AP if it was previously
  // connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  //encButton.begin();
  //encoder.begin();
  encoder.setAccelerationEnabled(false);

  gfx.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  displaySplash();
  gfx.clearDisplay();

  //printStringln(0, 0, 1, WHITE, "Setup Completed");
  Serial.println("Setup Completed"); 

  delay(3 * 1000);
}

void loop() {
  static uint32_t lastService = 0;
  if (lastService + 1000 < micros()) {
      lastService = micros();
      encoder.service(); 
  }

  nav.doInput();
  if (nav.changed(0)) {//only draw if changed
    nav.doOutput();
    gfx.display();
  }
}

size_t printStringln(int16_t x, int16_t y, uint8_t s, uint16_t c,
                     const String& val) {
  gfx.setTextSize(s);
  gfx.setTextColor(c);
  gfx.setCursor(x, y);
  return gfx.println(val);
}

void displaySplash() {
  gfx.clearDisplay();
  gfx.drawBitmap(0, 0, Logo, 128, 32, WHITE);
  gfx.display();
  delay(2000);
}

result showEvent(eventMask e, navNode& nav, prompt& item) {
  Serial.print(F("event:"));
  Serial.print(e);
  return proceed;
}
```

closed 2020-01-28 by stale[bot]