ThingPulse/esp8266-oled-ssd1306 #161: Memory leak on reinit display

RestiveX, 2017-12-05:
An error in the library that causes a memory leak, in the case of a re-initialization of the display (for example, when the module is "hotplug" turned off or on).

For fix the bug:
In the file OLEDDisplay.h need be added string `uint8_t *buffer;` to string `uint8_t *buffer = NULL;` and string `uint8_t *buffer_back;` to string `uint8_t *buffer_back = NULL;`

Also, in the file OLEDDisplay.cpp, find the block:
`this->buffer = (uint8_t*) malloc(sizeof(uint8_t) * DISPLAY_BUFFER_SIZE);`
`    if(!this->buffer) {`
`      DEBUG_OLEDDISPLAY("[OLEDDISPLAY][init] Not enough memory to create display\n");`
`      return false;`
`    }`
and supplement it with
`  if(this->buffer==NULL) {`
`    this->buffer = (uint8_t*) malloc(sizeof(uint8_t) * DISPLAY_BUFFER_SIZE);`
`    if(!this->buffer) {`
`      DEBUG_OLEDDISPLAY("[OLEDDISPLAY][init] Not enough memory to create display\n");`
`      return false;`
`    }`
`  }`

and second block
`  #ifdef OLEDDISPLAY_DOUBLE_BUFFER`
`    this->buffer_back = (uint8_t*) malloc(sizeof(uint8_t) * DISPLAY_BUFFER_SIZE);`
`    if(!this->buffer_back) {`
`      DEBUG_OLEDDISPLAY("[OLEDDISPLAY][init] Not enough memory to create back buffer\n");`
`      free(this->buffer);`
`      return false;`
`    }`
`  #endif`
complete to
`  #ifdef OLEDDISPLAY_DOUBLE_BUFFER`
`  if(this->buffer_back==NULL) {`
`    this->buffer_back = (uint8_t*) malloc(sizeof(uint8_t) * DISPLAY_BUFFER_SIZE);`
`    if(!this->buffer_back) {`
`      DEBUG_OLEDDISPLAY("[OLEDDISPLAY][init] Not enough memory to create back buffer\n");`
`      free(this->buffer);`
`      return false;`
`    }`
`  }`
`  #endif`

palsbo, 2017-12-05:
Will it be changed in the download on GitHub?

<esp8266-oled-ssd1306@noreply.github.com>
Cc: "Subscribed" <subscribed@noreply.github.com>
Sendt: 05-12-2017 12:05:12
Emne: [squix78/esp8266-oled-ssd1306] Memory leak on reinit display 
(#161)

lorol, 2017-12-07:
@RestiveX - could you please check if https://github.com/squix78/esp8266-oled-ssd1306/issues/143 is somehow related to your findings?

@palsbo - there are 2 pages of issues and solution proposals, many of them are waiting for long time. IMO - if you need this change, just take it. It may take a long time if ever, to be consolidated.

RestiveX, 2017-12-08:
@lorol - As you can see above: I have minimally changed the author's code. I just added a NULL check before allocating memory (that is, if the memory used to be allocated before, it will not be allocated again).

closed 2019-12-04 by stale[bot]