ThingPulse/esp8266-oled-ssd1306 #162: Last ESP32 compiler make errors in previous warnings

jestebanes, 2017-12-14:
Hello, this issue appears with ESP32 Wemos Oled and the last compiler version

in OLEDDisplay.h line 253, the method must to return a value, I suggest the new line

virtual bool connect() {return true;}; 

in OLELDisplay.cpp in line 356, variable data must be used uninitialized, I suggest initialize to 0

uint8_t data=0;

in OLEDDisplay.cpp in line 381, enumeration value TEXT_ALIGN_LEFT not used, I suggest to introduce it

case TEXT_ALIGN_LEFT:
      break;  

in OLEDDisplayUi.cpp in line 387, variable posOfhighlightFrame must be used uninitialized, I suggest initialize to 0 in line 334

uint8_t posOfHighlightFrame = 0;

in OLEDDisplayUi.cpp in line 393, variable x, y must be used uninitialized, I suggest initialize to 0 in line 365

uint16_t x=0,y=0;

in OLEDDisplayUi.cpp in line 284, variable x, y, x1, y1 must be used uninitialized, I suggest initialize to 0 in line 254

int16_t x=0, y=0, x1=0, y1=0;

dsptech, 2018-01-07:
Hi,
same issues here. 
Applied the suggested fixes (but false on virtual connect to check if the overriding goes wrong) and all work fine.

jestebanes, 2018-01-07:
Ok thanks, @dsptech, I change the virtual connect to return false.
thanks for the improvement.

closed 2019-12-04 by stale[bot]