ThingPulse/esp8266-oled-ssd1306 #165: Change pins I2C and library BMP085 Adafruit

sabas1080, 2017-12-20:
Hi @squix78 

I have a problem using the library oled and [BMP180 Adafruit](https://github.com/adafruit/Adafruit_BMP085_Unified) with this [ESP32 Board](https://www.banggood.com/Wemos-TTGO-WiFi-Bluetooth-Battery-ESP32-0_96-Inch-OLED-Development-Tool-p-1213497.html?rmmds=myorder&cur_warehouse=CN), the pins oled are 4 and 5 work standalone, but added library BPM085 not work BMP180, BMP180 work standalone in pins 21 and 22 (I2C default ESP32)

Do you know any way to make this work?

Thanks

ronalLeiva, 2017-12-23:
Hi @sabas1080 have you found a solution? I am trying to connect a BMP280 but when I operate the screen, the sensor stops working on the I2C bus. I have searched for information but I can not find how to solve

sabas1080, 2017-12-23:
Hi @ronalLeiva no yet

sabas1080, 2017-12-23:
I have a issue also in adafruit forum https://forums.adafruit.com/viewtopic.php?f=24&t=128358&sid=87f2c6e24153a1c873c0fb1d33bbb588

lexus2k, 2018-02-15:
Hi Everybody,
I didn't dig deep in the source code of this library, but looking at the place, where it initializes i2c, I found strange place:
 ```
     Wire.begin(this->_sda, this->_scl);
      // Let's use ~700khz if ESP8266 is in 160Mhz mode
      // this will be limited to ~400khz if the ESP8266 in 80Mhz mode.
Wire.setClock(700000);
```
Try to reduce clock speed for i2c. And remember that many devices doesn't support speed over 400kHz, and some support only 100kHz i2c.
Also, I think you need to update libraries for your sensor and OLED to have Wire initialization in single place.

01ivr3, 2019-08-13:
Does anyone know of a solution on how to use this ssd1306 library simultaneously with a second i2c device connected to an ESP32?! 

Note: there is another open issue #235 similar to this one.

Sylver11, 2019-11-23:
I assume there is still no fix for this? Also tried to change clock speed for I2C in the /libraries/esp8266-oled-ssd1306-master/src/SH1106Wire.h file but does not fix the problem...

closed 2020-06-05 by stale[bot]